ESPWortuhr/Multilayout-ESP-Wordclock #425: Wetteranzeige

MVarious, 2024-08-18:
Hallo, zu erst ein Dankeschön für dieses tolle Projekt.
Ich wollte mir nun die Variante mit dem Wetter bauen, doch leider bin ich auf ein paar Probleme gestoßen.

Hat jemand schon mal solch eine Uhr zum laufe bekommen. Ich wäre sehr daran interessiert was ich falsch gemacht habe.
Bei der normalen Uhr ist es kein Problem 4 Leds unten für die Sekunden einzusetzen..
Bei der WetterVariante fängt es dann leider an zu spinnen. auch die Einstellmöglichkeiten in dem Webinterface helfen nicht. 

Momentan "fehlen mir" 3 leds.. damit ich auch noch zum Ende das °C angezeigt bekomme.. Scheinbar funst alles bist auf die letzten beiden Zeilen. das Wort ZWANZIG ist zu kurz "beleuchtet" und damit ist das DREIßIG verschoben und das °C ist nicht mehr auf dem Raster. 

Ich wäre über ein paar tipps sehr dankbar

Liebe Gruß
Michael

dbambus, 2024-08-18:
Hey @MVarious,

mich schaue mir morgen mal das Problem genauer an ;-) aber kannst du mal ein Foto der Uhr machen ? Wie viele LEDs sind für die Word Matrix da, die Sekunden und die Wettermatrix ?

Mit freundlichen Grüßen
David

MVarious, 2024-08-18:
Hallo, klasse für die schnelle nachricht.. Ich habe in der Zwischzeit verschiedene Versionen durchgetestet.. aber bei allen scheinbar der gleich fehler.. was mir auch aufgefallen ist, das wenn ich die Wetter API eingebe.. sie nach dem Seitenrefresh nicht mehr sichtbar ist.. Ist das nur Datensicherheit oder ein Fehler?
Bilder sehen so aus. also die Uhr oben klappt prima. die letzten 2 Zeilen machen mir Sorgen.
Und wie gesagt.. egal was er anzeigt,, das °C fehlt immer.. hänge ich jedoch an die letzte Led noch welche dran, scheint das angezeigt zu werden nur irgendwie nach hinten raus geschoben.

Danke fürs anschauen
[image]
[image]
[image]

dbambus, 2024-08-21:
Hey, 

ich hab das grad überprüft und mir paar Haare rausgerissen. Es fehlt eine Zeile von Buchstaben. Ich habe das mal in dem Commit hinzugefügt. 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/compare/main...dbambus:Wortuhr:main 

Ich habe selbst diese Uhr nie gebaut, deswegen ist mir der Fehler nicht aufgefallen. 

Mit freundlichen Grüßen
David

MVarious, 2024-08-31:
Hallo an die Runde
Ich habe im Chancelog von der Beta 3.3 nicht rauslesen können, das es eine Änderung für das Wetterproblem gab, habe sie dennoch aufgespielt und leider sind die Probleme mit dem verschobenen zwanzig und dreißig weiterhin vorhanden. Und weiterhin sind die LEDs die das ° Symbol abdecken aus der Matrix "rausgeschoben" wenn man an die letzte Led der Matrix noch weitere LEDs dranbaut sieht man diese dann leuchten. Beste Grüße Micha

MVarious, 2024-09-15:
Hallo, Ich habe eben die neuste Beta auf meiner Wort/Wetter Uhr aufgespielt.. Die Überarbeitung des Webinterface sieht klasse aus. Leider bleiben die Probleme mit den letzten Zeilen bestehen.. Das Wort ZWANZIG ist verschoben und °C ist weiterhin außerhalb der Matrix. Das Wort Zwanzig wird nach einigen Sekunden mit etwas anderem ausgetauscht was für mich erstmal keinen Sinn macht. Ich habe die Uhr im Regenbogen und allen anderen Effekten laufen lassen. Dort sind alle LEDs korrekt an... Hier ein Video 

https://github.com/user-attachments/assets/de959d62-e6aa-4c93-805c-c72eedceaf77

closed 2024-09-18 by MVarious