ESPWortuhr/Multilayout-ESP-Wordclock #435: Add new customized layout

nikolai2111, 2024-09-08:
Hello everyone,

I am currently working with my colleague on a special customized version of the clock, the layout should not go into the official repo but only work in my fork.

Unfortunately I have not found a chapter in the documentation to add a customized layout manually, is there a possibility that someone who knows the code in detail could write here how and which adjustments would have to be made?

dbambus, 2024-09-08:
Hello,

Can you specify the layout a little bit?  Which Language? Matrix size? Standard or LED skipping for the arrangement?

Basicly take a look in the „Uhrtype“ Folder. Every .h/.hop definition you’ll add there will be available as header in the code as soon as you start the platform.io building process. There you can add new types of layout.

For a almost standard clock (10x11, DE, Standart LED) you can take a look at DE.10x11.Alternative.hpp 

If you have an other matrix configuration you’ll need to use the override functions. 

Mit freundlichen Grüßen
David 

PS: Just a question, why don’t you want to share your layout ;-)

nikolai2111, 2024-09-08:
Hello,

It's a CH10x11, but with some custom and personal jokes. It's not that I want to hide it, it's also publicly visible in my fork (you can [have a look](https://github.com/nikolai2111/Multilayout-ESP-Wordclock/tree/JoergUhr)), but it doesn't make sense to include it by default in the layout collection because it doesn't really make sense for others.

Currently I have copied and renamed CH10x11.hpp, in which I have modified the `setFrontMatrixWord()` function calls to control the correct LEDs. Are there any other changes that need to be made to the rest of the code?

Do I just need to save the layout as an .hpp file and it will be added directly to the collection? I thought I would have to add the drop down menu to select the layout etc.

I'm also planning to add a branch with a PCB to my fork, which may be more relevant to this project than the custom layout. I have a certain idea / vision. ;P

nikolai2111, 2024-09-08:
Currently, I modified the following parts:
1. Added a new header file called `CH10x11.custom.hpp`. In this file I created the class `Ch10x11Custom_t` and mapped the LEDs. In the last line I also defined the pointer `Ch10x11Custom_t _ch10x11custom;`
2. Added the following entry in the `ClockType` enum in the `Uhr.h` file:
```cpp
Ch10x11Custom = 99
```
3. Added the following entry in the switch-case-statement in the `clockWork.hpp` file:
```cpp
case Ch10x11Custom:
    return &_ch10x11custom;
```

I think I've forgotten something, even with the search function I couldn't find more. But currently I can't select my custom version in the drop down menu.

dbambus, 2024-09-09:
I took a look in your current fork ist looks okay from my side. Does ist build and work ?

closed 2024-09-09 by dbambus

nikolai2111, 2024-09-11:
Yes, it runs perfectly fine. Thanks!