ESPWortuhr/Multilayout-ESP-Wordclock #445: Einige Files werden nicht mehr erzeugt bei komplett neuem clone.

RoHA1965, 2024-09-19:
Bis zum b78d0dbba1c2c1bc62db5dd892e1112db47949ee (und ggfs etwas darüber) kann man neu clonen und bilden. Darüber kann man nicht mehr komplett neu bilden.
Es fehlt dann z.B. WebPageContent.gen.inc. Diese wird nicht mehr gebildet. Wenn sie von einem vorherigen build bereits erstellt wurde, geht es natürlich (z.B. nach einem pull).

dbambus, 2024-09-20:
Hallo @RoHA1965,

ich habe es grade bei mir getestet. Die Software baut so wie sie es soll. probiere mal einen Platformio Clean auszuführen und danach ein neuen Build Prozess auszulösen. 

Mit freundlichen Grüßen
David

RoHA1965, 2024-09-21:
clean hat keine Änderung gebracht. Ich habe es auch auf einem brandneu aufgesetzten Kubuntu Rechner versucht. Genau das gleiche. Ich habe es noch ein bisschen genauer eingegrenzt. Ab 296b7167ae4196b27efde86e6e3b3d24280b6160 geht kommt der Fehler.
Ich will auch noch mehr vergleichen, vielleicht finde ich den Grund.

git checkout 296b
run pio 
.
.
.
Compiling .pio/build/nodemcuv2/lib0dd/ESP8266WiFi/WiFiClient.cpp.o

Running "stylelint:all" (stylelint) task
Warning: Unexpected token '.' Use --force to continue.

Aborted due to warnings.
*** [include/WebPageContent.gen.inc] Error 3

RoHA1965, 2024-09-21:
Welche Buildumgebung verwendest Du?

Ich verwende Linux Kubuntu 22.04. (Auch auf dem neu aufgesetzten PC)
Ich hatte eben das Vergnügen es auf einem Windows 11 Rechner zu kompilieren. Da konnte ich es durchkompilieren mit der neusten Version !?

RoHA1965, 2024-09-26:
Es lag an node.js. Da wird jetzt wohl version >= 18 benötigt. Kubuntu hatte noch eine ältere drauf.
Das sollte evtl. bei der nächten Readme mit angegeben werden.

closed 2024-09-26 by RoHA1965