ESPWortuhr/Multilayout-ESP-Wordclock #450: 430 LEDs habe ich verbaut benötige hilfe bei der Umstellung des Codes

Morpheus2510, 2024-09-30:
Hallo zusammen,

ich habe die Uhr auf einer Fläche von 1x1 Meter nachgebaut, bekomme es aber leider nicht hin, dass 3 LEDs pro Buchstabe leuchten. Auch die restlichen Funktionen funktionieren nicht wie gewünscht. Wer kann mir bei der Umsetzung behilflich sein?

Danke im Voraus!

"LEDs (Reihe 1)": ["1–3", "5–7", "9–11", "13–15", "17–19", "21–23", "25–27", "29–31", "33–35", "37–39", "40–43"],
"LEDs (Reihe 2)": ["86–84", "82–80", "78–76", "74–72", "70–68", "66–64", "62–60", "58–56", "54–52", "50–48", "46–44"]

dbambus, 2024-09-30:
Hey @Morpheus2510,

also das Problem besteht aus zwei kleineren Paketen die gelöst werden müssten.

 > Leuchten von LEDs in anderen Rasterabstufungen.

Da muss nur eine Funktion geändert werden. Die `getFrontMatrixIndex` in Uhrtype.hpp und da an der Implementierung von DoubleResM1 orientieren. Da würde ein *4 besser passen ;-) 

 > Leuchten von mehreren LEDs pro Buchstabe

Das ist noch nicht implementiert und ich hab mir noch keine schöne Lösung für das Problem überlegt, ohne das alle Funktionen dann auch kaputt gehen.Aber das wird demnächst sicherlich kommen. Daher noch etwas Geduld.

Ich hoffe das kann helfen.

Mit freundlichen Grüßen
David

Morpheus2510, 2024-10-04:
@dbambus 

Hi David,
ja, zum Glück sind das nur zwei Baustellen. :-)
Ich weiß jetzt, was du meinst. Ich versuche das seit zwei Tagen zu lösen und erhalte immer mehr Fehler!

dbambus, 2024-10-12:
Hallo @Morpheus2510 

Bleiben wir mal bei Baustelle 1 :-) 

Welche Fehler kommen denn ? 

Mit freundlichen Grüßen
David

emilstrehl, 2024-10-14:
Hallo,

@Morpheus2510 
habe in meiner Uhr zwei LEDs pro Buchstabe und dazu wurden die zwei LED Stripes parallel angeschlossen.
Mit drei sollte es auch gehen, wenn das Netzteil die Spannung hält.
Ist bei dir vielleicht auch möglich und muss dann nicht extra Programmiert werden.

gruni2510, 2024-10-15:
[image]

[image]

So sieht die Uhr aus, habe jetzt schnell neuen Code geschrieben

gruni2510, 2024-10-15:
[image]

[image]

So schaut mein Projekt aus 

Habe jetzt einen passenden Code geschrieben zwar einfach aber funktioniert

Morpheus2510, 2024-10-15:
So sieht mein Projekt aus 1x1 Meter 

Code habe ich jetzt schnell selbst geschrieben das die Uhr wenigstens funktioniert 

[image]

[image]