ESPWortuhr/Multilayout-ESP-Wordclock #452: Farbreihenfolge anpassen

emilstrehl, 2024-10-06:
Hallo zusammen,
vielen Dank für dieses Projekt.
Ich habe mir auch so eine Uhr gebaut und etwas übertrieben mit 130x130cm.

Als LEDs habe ich von AliExpress die Weihnachtsbeleuchtung Kette genommen mit WS2812 LEDs. Bei der Größe mussten aber zwei parallele geschaltet werden.
Aber funktioniert super.

Bei diesen LEDs passt aber die Farbreihenfolge nicht, benötigt wird BGR( Blau Grün Rot).
Auf der Konfigurationsseite gibt es leider nicht passenden.

Kann mir vielleicht jemand ein Tip geben wo ich es verändern kann?

dbambus, 2024-10-06:
Hey @emilstrehl ,

man kann es auf der Website einstellen unter `Einstellungen > LED Strip` da gibt es die Auswahl zwischen WS2812 GRB, RGB, … 

Hier einfach das passende auswählen ;-) 

Mit freundlichen Grüßen
David

emilstrehl, 2024-10-06:
Hi @dbambus ,

vielen Dank für den Hinweis.
Die Möglichkeiten dort haben ich schon ausprobiert und aus den 4 Varianten ist keine passend.

Gruss
Emil

dbambus, 2024-10-12:
Hallo Emi, 

Ich habe das mal mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/458 behoben :-) 

Ein kleiner Aufruf zu Mithelfen. Ich habe unter https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/blob/main/docs/Anleitung.tex eine Latex Datei bereitgestellt in der ich Versuche eine Anleitung für die Uhr zu erstellen. 

Mit freundlichen Grüßen
David

closed 2024-10-12 by dbambus