ESPWortuhr/Multilayout-ESP-Wordclock #455: WLAN probleme bei "NO_TRANSITION"

RoHA1965, 2024-10-11:
Oft gibt es Probleme mit der WLAN Verbindung, wenn man auf "keine" stellt. Bei einem Restart ist dann oft Minutenlang kein Ping möglich. Oft geht es nach ein paar Minuten wieder, und manchmal hilft auch nur ein Powercycle und zusätzliches Warten. Das habe ich  NUR bei "keine" Transition. Bei "Überblenden" und andere (Fire geht ja generell gerade nicht) gibt es keine Probleme.
Ich habe schon versucht zu analysieren, aber ohne Erfolg.

dbambus, 2024-10-12:
Hallo @RoHA1965,

leider kann ich den Fehler nicht genau reproduzieren. Kannst du evtl einen Serial Log dann erstellen ? 

Mit freundlichen Grüßen
David

RoHA1965, 2024-10-13:
Neu gepullt auf f05d7c1.
Firework geht. 
Das WLAN hängen kommt nicht immer. Jetzt dachte ich schon es wäre weg, aber beim 4. "Nei Starten" über menu mit parallel laufendenm Ping, hing der Ping ca. 1 Minute. Auf der Seriellen kommen aber im Sekundentakt die Zeit.
Ich meine, dass es schlimmer wird, wenn man nach dem "neu initialisieren", Also nach neuem Eeprom schreiben durch ändern der SERNR, eine andere Transition Art anwählt und dann wieder zurück auf "KEINE" geht und neu started.
Ich kenne 2 weitere Personen, die das kennen.
Wie kann ich tiefere Logs erstellen über serial?

dbambus, 2025-02-21:
Hallo @RoHA1965,

besteht das Problem den weiterhin? 

Mit freundlichen Grüßen
David

RoHA1965, 2025-02-21:
Ja, aber ich nehme sowieso eine andere Einstellung. Ich glaube auch, wenn es auftritt, dann kriegt es sich nach längerer Zeit wieder ein.
Kann geschlossen werden, wenn es sonst niemand stört.

dbambus, 2025-02-21:
Alles klar :-)

closed 2025-02-21 by dbambus