ESPWortuhr/Multilayout-ESP-Wordclock #465: Sekunden...

MVarious, 2024-10-18:
Mir ist aufgefallen das ein alter Wunsch bezüglich der Hintergrundbeleuchtung ( Rahmenbeleuchtung) aufgegriffen wurde..

Wenn ich mich mit meinem Handybrowser ( der irgendwie ein Problem hat, da ich mich mit dem zwar auf die Website einloggen kann, aber er mir immer sagt das man Offline ist) zeigt er mir aber die Funktion Sekunden kann...

Diese kann ich aber leider nicht im Normalen PC Webbrowser finden.. 
Ich würde mich da über eine Erklärung freuen...

[image]

dbambus, 2024-10-18:
Hey,

ja das geht, aber nur beim Uhr 114 Alternative Rahmen. Da ist im Hintergrund entsprechender Code der das bei nicht unterstützen Konfigurationen ausblendet. 

Die Doku dazu fehlt allerdings 😅, das könnte man mal machen. 

Mit freundlichen Grüßen
David

MVarious, 2024-10-23:
Ah okay.,., wäre es möglich das hinter die anderen Uhrentypen auch zu legen? Leider kann ich in die Funktion nicht reinschauen.. ist das dann ein andere Pin vom ESP ( was sehr edel wäre ) das man da einfach ein paar Leds auf eine Farbe einstellen kann? Ich selber habe schon einige LED Uhren gebaut (Holzarbeiten) Mein Kumpel hat hat den Code geschrieben. Bei der Uhr hat er sich die Spielerei erlaubt das zu jeder vollen Stunde es eine Minute lang einen Regenbogenfarbwechsel gibt.. nur als kleine idee. Keine Ahnung ob das jemand anders auch mag.
https://github.com/merlinschumacher/esp8266-clock

P.S die Große Wortuhr mir Wetter habe ich in meiner Praxis hängen und sehr viele Kunden bleiben davor stehen und staunen.. Daher nochmal vielen dank für die Bereitstellung der Software. Tolle Arbeit!

dbambus, 2024-10-23:
Hallo @MVarious,

in der Software ist es (bisher) so gelöst, dass der LED-Streifen in 3 Teile {MATRIX}{MINUTEN}{SEKUNDEN/RAHMEN} aufgeteilt wird diese drei Teile müssen aufeinander folgen. Also 110 LEDs für die Matrix, dann 4 für die Minuten und dann einige für den Rahmen. 

Noch ist der Code nur für die Sekunden geschrieben. Ein Ambilight soll folgen. Andere Uhrtypen kann man relativ einfach umbauen, dazu muss man diese Zeilen Code in den Uhrtypen hinzufügen und für sich anpassen. Danach sollte es funktionieren :-) 

```
    virtual inline const uint8_t numPixelsFrameMatrix() { return 48; }

    //------------------------------------------------------------------------------

    virtual const bool hasSecondsFrame() override { return true; }

    //------------------------------------------------------------------------------

    virtual const uint16_t getFrameMatrixIndex(uint16_t index) override {
        return 114 + index;
    };
```

Aber ich finde die Lösung mit einem anderen Pin nicht schlecht. Ich hatte es nicht auf dem Schirm, neben der DMA-Methode (die ich benutze und wo der Pin leider auf "RX" gesetzt ist) auch eine andere Methode für den LED-Streifen zu verwenden. Das ermöglicht einige neue Anwendungsszenarien :-)

Super cool, dass du die Wetteruhr gebaut hast, könntest du bei Gelegenheit mal ein schönes Foto davon machen, das man evtl. als Beispiel hier zeigen könnte?

MVarious, 2024-10-25:
Hey, das klingt ja super, Freue mich auf die Umsetzung. Ich habe diese Bilder hier griffbereit. Sag mir einfach ob sie dir zusagen. Wenn nicht dann kann ich noch ein neues machen. 
Gibt es eigentlich eine Möglichkeit die Hintergrundfarbe zu verändern?

[image]

[image]

MVarious, 2024-10-25:
Eine kleine Schreibtischvariante habe ich auch gebaut.. hat ne richtige Sucht ausgelöst.. Meine Freundin hat eine Version hinter einem Spiegel bekommen. 
Ach, einfach herrlich sone Spielerei!!!

[image]
[image]
[image]
 und hier die Spiegelversion
[image]
[image]

dbambus, 2026-10-23:
Hallo @MVarious,

vielen Dank für das Issue und die tollen Fotos deiner Uhren, die Wetteruhr in der Praxis und die Spiegelversion sehen super aus! Der Sekundenrahmen ist ab 5.0 nicht mehr an „Uhr 114 Alternative Rahmen“ gebunden, sondern eine allgemeine Funktion für alle Layouts (außer 16x8 und dem Wetter-Layout). Unter Anzeigeoptionen → Sekunden stellst du die Anzahl der LEDs im Rahmen ein (0 = deaktiviert) und wählst Punkt, Sektor oder umlaufenden Sektor. Die Rahmen-LEDs hängen wie bisher am selben Streifen hinter der Matrix und den Minuten-LEDs. Die Farben für Vordergrund, Hintergrund und Rahmen lassen sich im Webinterface getrennt einstellen.

Enthalten im Release Candidate V5.0.0-rc.1:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V5.0.0-rc.1

Achtung: Mit dem Update werden alle Einstellungen zurückgesetzt.

Ich schließe das Issue. Für die Idee mit dem Regenbogen zur vollen Stunde mach gerne ein eigenes Issue auf.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus