ESPWortuhr/Multilayout-ESP-Wordclock #469: Modus und Laufschrift per MQTT ändern

Pauli29, 2024-10-24:
Ist es eigentlich möglich den Modus und die Laufschrift per MQTT zu ändern?

dbambus, 2024-10-24:
Hallo @Pauli29,

Das ist möglich, in den Zeilen 209ff in der MQTT.hpp ist es hinterlegt:

```
    const char *effect = doc["effect"];
    if (doc.containsKey("effect")) {
        if (!strcmp("Wordclock", effect)) {
            G.prog = COMMAND_MODE_WORD_CLOCK;
        } else if (!strcmp("Seconds", effect)) {
            G.prog = COMMAND_MODE_SECONDS;
        } else if (!strcmp("Digitalclock", effect)) {
            G.prog = COMMAND_MODE_DIGITAL_CLOCK;
        } else if (!strcmp("Scrollingtext", effect)) {
            G.prog = COMMAND_MODE_SCROLLINGTEXT;
        } else if (!strcmp("Rainbowcycle", effect)) {
            G.prog = COMMAND_MODE_RAINBOWCYCLE;
        } else if (!strcmp("Rainbow", effect)) {
            G.prog = COMMAND_MODE_RAINBOW;
        } else if (!strcmp("Color", effect)) {
            G.prog = COMMAND_MODE_COLOR;
        } else if (!strcmp("Symbol", effect)) {
            G.prog = COMMAND_MODE_SYMBOL;
        }
    }

    // Copy marquee_text if present
    if (doc.containsKey("marquee_text")) {
        strcpy(G.scrollingText, doc["marquee_text"]);
    }

```

Beispiele wären:

{
    "effect": "Scrollingtext",
    "marquee_text": "Hallo Welt"
}

Mit freundlichen Grüßen
David

dbambus, 2024-10-27:
Hallo @Pauli29, 

ich habe noch zwei Bugs im MQTT entdeckt und gefixed und nebenbei noch die `Mqtt.hpp` mit Dokumentation ausgestattet. 

Im selben Zuge habe ich die Scrolling Text etwas glattgezogen. Sodass es jetzt 

```
{
"effect": "Scrollingtext",
"scrolling_text": "Hallo Welt"
}
```

sein wird. 

@Pauli29 ich habe mir das mal als ToDo in die `MQTT.hpp` hinterlegt. Aber ehrlich gesagt verstehe ich den Sinn noch nicht so ganz. Könntest du evtl. ein config Beispiel for Homeassistant geben mit den punkten die wichtig wären ?

Mit freundlichen Grüßen
David

pfefferle, 2024-10-28:
@dbambus ich glaube du meinst mich :)

`Notifications` sind, genau wie `Lights` ein Entity-Typ in Home Assistant und für das Versenden von Benachrichtigungen gedacht: https://www.home-assistant.io/integrations/notify/

Ich dachte, dass das ja genau zu "Scrollingtext" passen würde.

dbambus, 2024-11-12:
hey @pfefferle,

ich werde mir das mal notieren, als neues Feature. Aber dieses Issue hier schließen, da es nicht allzu gut zur Überschrift passt. 

Mit freundlichen Grüßen
David

closed 2024-11-12 by dbambus