ESPWortuhr/Multilayout-ESP-Wordclock #476: Buchstaben Layout

J-Paul0815, 2024-11-05:
Bin gestern auf dieses Projekt gestoßen und bin schon sehr begeistert. Ich hoffe ich stelle meine Frage an richtiger Stelle, ein Forum oder Discord scheint es nicht zu geben. Bei mir ist das Buchstaben Layout anders, siehe Bild. Ich hatte ein Issue gefunden, in dem es hieß, dass man alternative Layouts hochladen kann, ich finde aber nicht was und wie. Hier das Layout:
[image]
Wäre über jeden Hinweis sehr dankbar.

dbambus, 2024-11-05:
Hallo @J-Paul0815 

auf der Website der Uhr kann man zwischen mehreren Layouts wählen. Dein Design entspricht dabei dem Layout `2clock`

andernfalls in der Config diese Zeile auskommentieren. 

```
// #define DEFAULT_LAYOUT Ger10x11Clock
```

Falls du weitere Hilfe braucht oder wirklich ein eigene Layout haben willst, dann nochmal melden. 

Mit freundlichen Grüßen
David

J-Paul0815, 2024-11-05:
> Hallo @J-Paul0815
> 
> auf der Website der Uhr kann man zwischen mehreren Layouts wählen. Dein Design entspricht dabei dem Layout `2clock`
> 
danke @dbambus für die schnelle Antwort.
Wo finde ich ich die Auswahl vom Layout? Ich hatte mit der BIN Version 4.0.2 geflashed, fehlt mir da die "Experten" Einstellung?
LG Joerg

closed 2024-11-12 by dbambus