ESPWortuhr/Multilayout-ESP-Wordclock #479: Timer Rest Minuten Anzeige

J-Paul0815, 2024-11-11:
Bin nach wie vor Begeistert von diesem Projekt, nicht zu vergleichen mit meiner selbst gestrickten Version, vielen Dank!

Was bei meiner selbst gebauten Version allerdings als Funktion hatte, war die Anzeige der Rest Minuten eines (Alexa) Timers, gesteuert über MQTT. Über per MQTT gesteuerte Funktion:
```
{
  "effect": "Scrollingtext",
  "scrolling_text": "37 "
}
```
kann man ja eine Rest Laufzeit schicken, anzeigen lassen, schön wäre allerdings eine statische Anzeige wie beim Modus "Sekunden".
Wäre es möglich 
1. Bei Laufschrift die Geschwindigkeit auf 0 zu setzen und
2. Die Änderung der Geschwindigkeit auch per MQTT einstellbar zu haben?

Vielleicht übersehe ich auch etwas, so habe ich es nicht gefunden, wie per MQTT was gesteuert werden kann.
Gibt es eine Anleitung, oder Übersicht?

dbambus, 2024-11-12:
Hey, 

das hört sich nach einer super Änderung an, die ich demnächst implementieren werde. Ebenfalls ein cooler Einsatzzweck :-) 

Mit freundlichen Grüßen
David

dbambus, 2024-11-13:
A few moments later xD

Ich werde mal sehne wie ich das besser implementieren kann :-) 

Ich stelle mir da etwas vor wie, eine If-Abfrage die prüft, ob nur bis zu zwei Zeichen kommen und wenn ja, dann werden diese Zeichen ohne shift speed dargestellt. Wie die Sekunden Anzeige. 

PS: Wie hast du die Rahmen animation gelöst ? Hast du dazu einen Fork ? 

Mit freundlichen Grüßen
David

J-Paul0815, 2024-11-14:
Moin David,
Dass mit der Logik, dass es bei <3 Zeichen statisch und ab 3 Zeichen als Laufschrift hatte ich mir auch gedacht, hätte das aber genau wie die Logik zur Berechnung der Restminuten vom Timer in mein (Blockly-) Skript in Iobroker integriert. 
Die Umrandungsbeleuchtung kommt von einem separaten Wemos mit WLED. Neben der Animation zum Timerende soll zwischen Sonnenuntergang und Sonnenaufgang als statische Hintergrundbeleuchtung dienen, in der gleichen Buchstabenfarbe, die stündlich wechselt.

dbambus, 2026-10-23:
Hallo @J-Paul0815,

vielen Dank für die Idee und das Video, dein Anwendungsfall hat mir sehr gefallen. Umgesetzt ist es so, wie wir es besprochen hatten: Besteht die Laufschrift aus höchstens zwei Zeichen, wird sie nicht mehr gescrollt, sondern statisch angezeigt, zwei Ziffern erscheinen wie im Sekundenmodus. Leerzeichen am Anfang und Ende werden ignoriert, dein Beispiel funktioniert also direkt:

```
{
  "effect": "Scrollingtext",
  "scrolling_text": "37 "
}
```

Die Geschwindigkeit lässt sich zusätzlich per MQTT setzen (`<TOPIC>/effect_speed/set`, Werte 1–10), in Home Assistant erscheint sie als Entität „Effect Speed“. Eine Übersicht der MQTT-Befehle steht in der README im Abschnitt „MQTT API Dokumentation“.

Die statische Anzeige ist seit V4.4.0 dabei, aktuell ist der Release Candidate V5.0.0-rc.1:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V5.0.0-rc.1

Ich schließe das Issue.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus