ESPWortuhr/Multilayout-ESP-Wordclock #482: Animations not mirrored

MajorTwip, 2024-11-12:
If you select to mirror the LED (in my case horicontally and vertically which is in fact a rotation) the animations are not mirrored.

Especcially the matrix-rain falling upwards is a pity.

dbambus, 2024-11-12:
Hey @MajorTwip,

thank you for the Bug report. I will take a look into it. 

Cheers 
David

dbambus, 2026-10-23:
Hey @MajorTwip,

thanks again for reporting this, and sorry that it took so long. The transitions are now calculated in display orientation, so with horizontal and/or vertical mirroring they run the right way round again – the matrix rain falls down instead of up. The fix came with #809 (thanks @timostr) and is part of the release candidate V5.0.0-rc.1:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V5.0.0-rc.1

Please note that all settings are reset with this update, so write down your configuration before flashing.

I'll close this issue. If anything still runs the wrong way, just reopen it.

Cheers
David

closed 2026-10-23 by dbambus