ESPWortuhr/Multilayout-ESP-Wordclock #487: Animantion alle 5 Minuten läuft auch wenn wenn Helligkeit auf "Aus"

jodah30, 2024-11-23:
Bei Release 4.0.2 habe ich die Uhr so eingestellt: 0-6 Uhr Helligkeit auf "Aus"
Wenn allerdings die Annimation "Laser" startet, ist diese trotzdem sichtbar.

Tolles Projekt! Habe große Freude daran

dbambus, 2024-12-06:
Hallo @jodah30,

vielen Dank für deinen Bugreport, ich glaube auch schon zu wissen wo der Fehler ist. Es sollte sich also leicht beheben lassen. 

Mit freundlichen Grüßen
David

dbambus, 2026-10-23:
Hallo @jodah30,

vielen Dank für dein Issue und sorry, dass es so lange gedauert hat. Die Ursache war genau da, wo ich sie vermutet hatte: Der Laserstrahl war fest auf volles Weiß gesetzt und das Feuerwerk nutzte feste, volle Farben. Beide haben die Helligkeit gar nicht beachtet – nicht nur bei „Aus“, sondern auch bei jeder gedimmten Stufe in der Nacht. Das galt auch für das Geburtstags-Feuerwerk.

Mit #828 übernehmen Laser und Feuerwerk jetzt die aktuelle Helligkeit der Uhr. Damit folgen sie dem Zeitplan, der automatischen Helligkeit und dem Aus-Zustand. Bei 100 % sehen sie aus wie bisher, bei 10 % sind sie so schwach wie die Wörter, und bei 0 % bleibt die Uhr komplett dunkel. Ich habe das auf einer echten Uhr getestet.

Der Fix kommt mit der nächsten Version. Ich schließe das Issue; falls du danach noch etwas aufblitzen siehst, öffne es gerne wieder.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus