ESPWortuhr/Multilayout-ESP-Wordclock #488: M5 Atom Lite ESP32

irqnet, 2024-11-25:
Dank dem Port auf den ESP32 gibt es von mir eine neue Controller Empfehlung für das Wortuhr Projekt!

M5 Atom Lite
- ESP32 Board
- 25x25x10mm

Getestet mit dem aktuellen Branch, Anpassung der GPIO Ports.

Der Vorteil, das Modul hat einen Grove Connector, kann darüber mit Spannung versorgt werden und G26 wird darüber auch nach außen geführt.

Ich warte noch auf meine Grove Stecker, aktuell hab ich es getestet mit den PIN Outs an der Oberseite und G25 als LED GPIO.

Link: https://shop.m5stack.com/products/atom-lite-esp32-development-kit?srsltid=AfmBOorJlBeVEc20sRT6h21ZlERPMxOh7-uQDucpkZR7-prCg-rLMWwx

[image]

[image]

irqnet, 2024-11-27:
Habe einen Test gemacht mit den Ports G26 und G32 die über den Anschluss unten verfügbar sind.
@dbambus 
Leider bekomme ich hier kein Ergebnis wenn ich den LEDGPIO entsprechend ändere. Unter WLED funktionieren beide Ports allerdings.

Gibt es hier noch irgendwo eine andere Stelle die im Code geändert werden muss?

dbambus, 2024-12-06:
Hey @irqnet,

ich muss mal sehen wie das spezifiziert ist. Ich verwende ja die Neopixelbus Library. 

Im Code selbst wird in der `clockwork.hpp` ein `#define LEDGPIO 27` gesetzt. Wie man der Schematic vom M5 ATOM entnehmen kann, ist da auch ein SK6812 angeschlossen, der sollte zumindest funktionieren.

Wenn man das `#define` aber abändert sollte es doch mit 26 bzw. 32 funktionieren, oder? Evtl muss man da tiefer in der verwendeten Lib nachschauen, was die Gründe dafür sind. Bspw. beim ESP8266 ist nur der RX Pin für die DMA Ansteuerung möglich. 

Mit freundlichen Grüßen
David

irqnet, 2024-12-09:
> Hey @irqnet,
>
> ich muss mal sehen wie das spezifiziert ist. Ich verwende ja die Neopixelbus Library.
>
> Im Code selbst wird in der `clockwork.hpp` ein `#define LEDGPIO 27` gesetzt. Wie man der [...]

Hi David,

das define habe ich entsprechend geändert aber es funktioniert mit den gpios am vorderen Anschluss eben nicht. Nur mit denen am breakout pin oben.

Kann natürlich sein das die Version der Neopixelbus Lib die ports nicht unterstützt? Aber beim kompilieren bekomme ich auch keinen Fehler.

Wüsste aktuell nicht wo ich da suchen muss.

irqnet, 2024-12-10:
Es scheint daran zu liegen das die GPIO Pins nicht default auf OUTPUT konfiguriert sind.

Hab bei der ESP32 Definition nun folgendes hinterlegt:

pinMode(LEDGPIO, OUTPUT);

irqnet, 2024-12-11:
@dbambus Ja konnte ich, wobei das jetzt ein erster Versuch war und ich die pinMode Definition sicher nicht an der richtigen Stelle mache.

Ich muss aktuell bei jedem Start/Neustart die LED Config neu auswählen, speichern, und zurück stellen, damit die überhaupt angehen. Vielleicht ist hier nochmal Hilfe nötig :)

[image]

[image]

Wie gesagt, das war ein Try and Error, aber so geht es mit den manuell beschriebenen Steps. Final ist das noch nicht :)

Irgendeine Idee wo man die GPIO Config für den ESP32 besser unterbringt?

[image]

irqnet, 2024-12-11:
Irgendwie scheint es aber nicht reproduzierbar zu sein. Mit einem zweiten Atom bekomme ich es nicht zum Rennen :/

dbambus, 2025-02-21:
Hey @irqnet,

ich habe mal deine PinMode Definition in das aktuelle Release gepackt. https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/508/commits/929b89d5a0c486a7c38cde523d6f30b995ec9573

Mit freundlichen Grüßen
David

closed 2025-02-21 by dbambus