ESPWortuhr/Multilayout-ESP-Wordclock #497: Misplaced WIFI logo on frst startup

nikolai2111, 2024-12-15:
I had an error with the WIFI logo at startup, only the first time when the flash is empty. After entering the WIFI credentials, the WIFI logo appeared correctly. When I cleaned the flash, the incorrect one reappears.

[image]

Sorry for the bad quality ;P.

PS: Yes, the first prototypes for the PCB have arrived! They look sooo clean, I was wondering if you would be interested in one?

dbambus, 2024-12-15:
Hey @nikolai2111, 

fantastic job :-) nice. I would be interested in one of those PCBs. 

Just for the recall. If you have a completely new code without any user inputs the Wifi Symbol is corrupted. As soon as you enter the credentials the Logo is just fine as long as you don't reset the clock again ? 

Can you activate the LEDSWEEP in the `config.h` to see if all the LEDs (except of the minute ones) light up correctly ? 

```
//--------------------------------------------------------------------------
// Settings for Boot sequence
//--------------------------------------------------------------------------
/*
 * Show Ip Adress of clock on boot, with BOOT_SHOWIP
 * Light up LEDs on boot, with BOOT_LEDSWEEP
 *
 * Valid values BOOT_SHOWIP [true, false]
 * Valid values BOOT_LEDSWEEP [true, false]
 */
#define BOOT_SHOWIP true
#define BOOT_LEDSWEEP true

```
Cheers 
David

nikolai2111, 2024-12-18:
Hello @dbambus,

How do we get in touch with each other?
I never did this before, so I have no clue other than [guerrillamail](https://guerrillamail.com). Because I don't want to put my direct email here. ;P

> Can you activate the LEDSWEEP in the `config.h` to see if all the LEDs (except of the minute ones) light up correctly ?
I did that, there should be no problem with that.

I checked it again and I had a very strange behavior.
After a reset and also a re-upload I did not have the problem, the WIFI logo shows up normal. I only had the error after I had programmed the empty and new ESP32, or after I manually erased the flash (with the "Erase Flash" button) and then re-uploaded the firmware (with the "Upload and Monitor" button).

dbambus, 2024-12-18:
hey @nikolai2111,

I have my real name in Github and you can find my email address relatively quickly :-) but here it is: 

david.panusch@fau.de

Just mail me. 

Ahh, I got it. You are using an ESP32. Well I just recently adapted the code to work on that MCU. There could be some strange behaviors, still. I will take a look into it to figure out what is going on here. 

Cheers
David

closed 2025-02-21 by dbambus