ESPWortuhr/Multilayout-ESP-Wordclock #498: Zeitzone einstellen

dieserkai, 2024-12-16:
Hallo,

gibt es einen Weg im Nachhinein nach dem kompilieren die Zeitzone einzustellen, oder bleib dort nur die Option zum neu kompilieren und OTA Update?

Ist geplant die Zeitzone künftig im Webinterface einstellbar zu machen?

dbambus, 2024-12-16:
Hey, 

In der Tat ist die Möglichkeit die Zeitzone zu ändern nicht implementiert. Es sollte aber eine relativ einfach Änderung sein, die das Ermöglichen könnte. :-) 

Aktuell ist es daher notwendig das im Code zu Ändern und neu zu kompilieren. Andernfalls kann man im Webinterface auch die Zeit manuell einstellen. 

Mit freundlichen Grüßen
David

dieserkai, 2024-12-16:
Das einstellen im Webinterface knallt dann halt beim ntp update. Also müsste man dann ach den ntp server entfernen.

Das OTA update per Webui funktioniert beim nodemcu v3? dann würde ich neu kompilieren und die bin weiterverteilen. die uhr ist in einer Zeitzone 9h hinter uns, also nicht wo man mal eben hingehen könnte.

Ist der Einbau der einstellbaren zeitzone in der WebUI zeitnah geplant (damit ich noch drauf warten kann), oder wird das noch ne weile dauern bis es kommt (falls es kommt)

Schöne Grüße,
Kai

dbambus, 2024-12-16:
hey @dieserkai, 

Das OTA sollte gehen :-) <IP-Adresse der Uhr>:81

Uff, theoretisch ist das schnell implementiert, aber ich bin grade im Urlaub auf reisen und testen kann ich das halt nicht. 

Mit freundlichen Grüßen
David

dieserkai, 2025-07-19:
@dbambus hattest du zwischenzeitlich ggf schon Zeit dir das mal anzuschauen? Ich muss bald noch eine Uhr für das Ausland fertig machen