ESPWortuhr/Multilayout-ESP-Wordclock #500: Schweizerdeutsch "gsi" fehlt

MajorTwip, 2024-12-17:
Das Layout in Deutsch Schweiz endet mit den Buchstaben "gsi". Übersetzt wäre dies "gewesen".
1415 wäre also "viertu ab zwöi", 1416 "viertu ab zwöi gsi". Die Logik entspricht somit eigentlich der ersten Minuten-LED.
Dies wäre noch ein Feature welches die Hardware bereits umsetzt.

dbambus, 2024-12-18:
Hallo @MajorTwip,

 ich habe grade in den Code geschaut. Prinzipiell einfach zu implementieren. Jedoch, beinhaltet das Layout `include/Uhrtypes/CH10x11.hpp` die Buchstaben GSI nicht? Ist da ein Fehler in dem Layout oder meist du ein kolpett anderes? 

Mit freundlichen Grüßen
David

nikolai2111, 2024-12-18:
Hallo zusammen,

Das wäre wohl eine Feature Request, da dies kein Fehler ist, sondern ein anderes Layout mit einer anderen "Bedeutung". Das wäre als würde man für 14:16 im Deutschen mit dem Wort "Viertel nach zwei gewesen" abschliessen.

MajorTwip, 2024-12-19:
[image]
Ihr habt recht, ich hab ein leicht anderes Layout.
Klar, dieses Issue ist als Feature Request zu werten, nicht als Bug.

MajorTwip, 2024-12-19:
Das GIF zeigt ausserdem das Issue #482

dbambus, 2025-01-14:
Hey @MajorTwip,

ich habe mal ein weiteres Layout CH10x11V2 hinzugefügt, dass das Wort "GSI" enthält. Die Logik dazu fehlt aber noch. Die kommt aber mit in das kommende Bugfix release. 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/508

Mit freundlichen Grüßen
David

dbambus, 2025-02-21:
Hey, 

ich wollte mich dem mal annehmen. Ich verstehe aber die Logik noch nicht ganz. 

Ist das "Gsi" bei der Minute die nicht Restlos durch 5 Teilbar ist? 

Oder 

Nur eine minute nach 5, 10, 15, 20, ... 

Mit freundlichen Grüßen
David

MajorTwip, 2025-02-22:
Hey
Ersteres ist korrekt.
"gsi" bedeutet "gewesen"
1520 = es isch zwänzg ab drü
1521-1524 = es isch zwänzg ab drü gsi

dbambus, 2025-02-22:
Hay @MajorTwip, 

ich habe es mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/533 hinzugefügt ^^

Mit freundlichen Grüßen
David

closed 2025-02-22 by dbambus

MajorTwip, 2025-08-28:
[image]
Du bist super !