ESPWortuhr/Multilayout-ESP-Wordclock #501: Kein 10 vor halb oder 10 nach halb

rolkow, 2024-12-19:
Ich verwende das Release 4.0.2 mit dem Standardlayout DE10*11 auf einem ESP8266.
In der WebUI kann man bei Sprache auswählen:
Um 9:20 ist es    -> zehn vor halb zehn
Um 9:40 ist es    -> zehn nach halb zehn

das funktioniert bei mir leider nicht - nur ein kurzes Zucken in der Anzeige - aber der Wortlaut ändert sich nicht

das Umstellen von  viertel nach neun auf viertel zehn funktioniert

Vielen herzlichen Dank für dieses tolle Projekt.

Grüße aus Bayern

Roland

dbambus, 2025-01-07:
Hallo @rolkow, 

ich bin erst jetzt wieder am PC. Werde mir das mal die Tage anschauen und dann einen Patch liefern. Vielen Dank für deinen Input. 

Mit freundlichen Grüßen
David

dbambus, 2025-01-14:
Hey @rolkow, 

da war tatsächlich ein Bug in der Software. Ich habe das mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/508 behoben. 

Mit freundlichen Grüßen
David

closed 2025-01-14 by dbambus