ESPWortuhr/Multilayout-ESP-Wordclock #509: Starten OHNE WLAN, nur RTC!?

Miklesoft, 2025-01-15:
Hallo, gibt es die Möglichkeit die Uhr OHNE WLAN, aber mit eingebauter RTC  zu betrieben?. Aktuell startet die Uhr nur wenn WLAN verfügbar!

Ansonsten tolles Projekt :)

dbambus, 2025-01-15:
Hallo, 

es funktioniert. Wenn eine RTC eingebaut ist, liest die Software beim Start die Uhrzeit aus und lässt das WLAN für 60 Sekunden eingeschaltet. Wird innerhalb dieser 60 Sekunden keine Verbindung aufgebaut, schaltet die Uhr das WLAN bis zum nächsten Neustart aus. 

Hinweis: 5 Schnelle Neustarts hintereinander führen zu einer Zurücksetzung auf "Werkseinstellungen".

Mit freundlichen Grüßen
David

Miklesoft, 2025-01-15:
Danke für die schnelle Antwort, War wohl zu ungeduldig:) Es klappt.
Gibt es denn auch die Möglichkeit auf die Weboberfläche zuzugreifen, Z.b. Über den AP der Uhr.

dbambus, 2025-01-15:
Hey @Miklesoft, 

leider noch nicht. Aktuell kann man nur mit einer aktiven WLAN Verbindung die Uhr einrichten. Sobald die Einstellungen gemacht wurden, kann man aber die Uhr auch ohne WLAN betreiben. 

Mit freundlichen Grüßen
David

closed 2026-03-02 by dbambus