ESPWortuhr/Multilayout-ESP-Wordclock #511: Number of Pixels for secods is limited to 64

Vrumfondel, 2025-01-15:
Number of Pixels for secods is limited to 64. I prefer to double this.
(From the pure display point of view 60 pixles should be enough, but one might run into the situation that there is no LED-stripes available one would need to nicely fill the room he has available for the seconds. Like in my case - I would need a stripe with 50LED/m, which is not availavle. So I want to use 100LED/m)

dbambus, 2025-01-15:
Hey @Vrumfondel, 

okay that could be changed fairly easy, do you want to try to change it and create a pull request ? 

Cheers 
David

Vrumfondel, 2025-01-15:
Yes, that's the plan.

dbambus, 2025-02-21:
Should be fixed with your PR https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/532

closed 2025-02-21 by dbambus