ESPWortuhr/Multilayout-ESP-Wordclock #512: LED Reihenfolge 424 Variante

Korny3005, 2025-01-16:
Hallo, danke für das Projekt. Hab viel Spaß damit.

Hast oder kannst du die Verdrahtungsreihenfolge der LEDs für die 424 Variante darstellen?
Mir ist nicht klar ob die 4 Minuten LEDs vor oder nach dem "Wetter-Teil" kommen müssen.

Danke

dbambus, 2025-01-16:
Hey @Korny3005, 

Die Verdrahtungsreihenfolge ist in abstrakter Form der Uhrtypedef-> DE22x11.weahter.hpp zu entnehmen. Hier kann man sehen, dass die Minuten mit den 4 LEDs als Reihe X (Real 11, da von 0 an begonnen wird zu zählen) definiert sind. 

So ergibt sich, dass man erst 110 LEDs (11x10) für die Wordclock ansetzt, dann kommen 4 einzelne für die Minuten und danach nochmal 110 LEDs (11x10) für die Wetteranzeige. :-) 

Ich würde mich freuen, wenn du mir von deinen Erfahrungen beim Bauen der Uhr erzählen könntest und wenn du mir sagen könntest, ob die Software einwandfrei funktioniert, da ich leider keine Uhr dieses Typs hier habe, um das zu testen. 

Mit freundlichen Grüßen
David

Korny3005, 2025-01-16:
Wird gemacht,
momentan läuft noch der Drucker auf hochtouren. Danke schonmal für die Info.

[image]

[image]

dbambus, 2025-01-17:
Hey @Korny3005, 

es sollte die im WLAN AP Modus nur die IP: 192.168.4.1 ohne Port sein. Der Port 1883 ist reserviert für das MQTT Protokoll. 

Auch hier kann es helfen über das esptool.py -erase_flash auszuführen und die Software nur auf die Uhr zu flashen. Ich habe selbst noch paar Probleme mit dem ESP32 controller und er funktioniert nicht immer in allen Fällen zuverlässig. Wenn es ein Geschenk werden soll, dann würde ich immer den ESP8266 empfehlen. 

Ich finde es sehr stark, wie du das 3D Gedruckt hast. Wirst du die SLT Daten eventuell zur verfügung stellen?

Mit freundlichen Grüßen
david

Korny3005, 2025-01-17:
hey @dbambus ,

die Dateien stell ich gerne in vollem Umfang zur Verfügung, dann kannst du es in das Projekt integrieren.
Sobald ich es zusammengebaut habe und die Teile auch ineinander passen, bekommst du diese gerne!
Incl. Druckbesonderheiten :)

Korny3005, 2025-01-17:
Yo @dbambus ,
ich habs hinbekommen. Genial.
Ich hab aber statt dem VIN die 3V3 Versorgung für die LEDs verwenden müssen.
Ach und dazu meinen ESP8266 controller.

Ich bin auch im WLAN und es läuft. 

Nur ist glaube ich, keine Variante vorgesehen für die, die ich verwendet habe.

[image]

Ist das für dich schnell umsetzbar? 
Wenn nicht würde ich evtl sogar meine Uhr umbauen.

2. Hast du nen Tipp für ne Wetter API. Gibts viele, einige sehr teuer, und bin mir nicht sicher welche funktionieren

Korny3005, 2025-01-19:
Ok break, ich hab gerade festgestellt, dass meine PSU nicht genug geliefert hat.
Ich setz nich die Tage nochmal dran!

closed 2025-01-25 by dbambus