ESPWortuhr/Multilayout-ESP-Wordclock #517: WLAN Probleme / kein Zugriff auf die Uhr

fs-sw, 2025-01-17:
Ich habe in verschiedenen WLAN Netzen das Problem, das ich die Uhr einbinden kann und dann nicht auf die Oberfläche via IP Zugreifen kann.
Die Uhr läuft dann aber ganz normal.
Bei Start wird auch keine IP Adresse angezeigt.
Wer hat eine Idee?
Danke
Frank

dbambus, 2025-01-17:
Hey @fs-sw,

erstmal zwei Fragen vorab. Welchen Mikrocontroller verwendest du? Hast du einen Seriellen Output von deiner Uhr?

Manchmal hilft es die Uhr über einen esptool.py -erase_flash zu löschen und dann neu zu bespielen. 

Mit freundlichen Grüßen
David

fs-sw, 2025-01-17:
Verwende diese Controller.
AZDelivery 3 x NodeMCU Amica Modul V2 ESP8266 ESP-12F WiFi - Node MCU ESP 8266 WiFi Development Board mit CP2102 kompatibel mit Arduino

Leider habe ich keinen Output, da dieses Phänomen nicht bei mir zu Hause auftritt.

Vrumfondel, 2025-01-17:
Nachdem ich mit meinem Boards auch sehr obscure Probleme mit wohl noch bestehendem Flashinhalt von einem vorhergehenden Projekt hatte: Wie wäre es eine Kennung zu hinterlegen ob der aktuelle Inhalt des Setting Bereiches wirklich von der Clock beschrieben wurde oder noch Altlasten sind, und wenn letzteres alles zu initialisieren?

fs-sw, 2025-01-17:
@Vrumfondel 
der hier?
ESP8266 NodeMcu Lua WiFi V3 Modul RUIZHI 2 Stück ESP8266 NodeMCU ESP12E WiFi Entwicklungsboard mit CH340 NodeMcu Lua WiFi 2.4 GHz V3 Modul kompatibel mit Arduino IDE/Micropython

[https://www.amazon.de/ESP8266-RUIZHI-Entwicklungsboard-kompatibel-Micropython/dp/B0CLY9B5MS/ref=sr_1_3_sspa?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=GQ5PDIKIK5B9&dib=eyJ2IjoiMSJ9.YQWcpkkr-aSRUb4YJqt2zD92iy30C9__VN3WsmUQWyrxLvHqGAEQeF35BZua4fonVjO6vfJciS-RL2_I7yIRymH3YoEmJ3ncxPp9A8vrodRTkoGdWklirf4U_73_OWUVzozYhiVTfhXTFesOAEPq-qPyMe82YgAKdOf89fLfI_nORCCdav0PC7IcL4VKQbWE83gJxGL3PqK0d8VKqw-iBUX8Qm7iYLgdyraSaYbRyHQ.V6SkON-LWAiULpxt3wKzl-b7iIjRkTfbJSFEaHEBlJQ&dib_tag=se&keywords=esp12e&nsdOptOutParam=true&qid=1737107165&sprefix=esp12f%2Caps%2C180&sr=8-3-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1](url)

dbambus, 2025-01-17:
Okay ein ESP8266. 

Wobei ich vorgestern echt mit meinem MCU zu kämpfen hatte. Nach einem -erase_flash hat es dann perfekt funktioniert.

Mit freundlichen Grüßen
David

Vrumfondel, 2025-01-17:
> [@Vrumfondel](https://github.com/Vrumfondel) der hier? ESP8266 NodeMcu Lua WiFi V3 Modul RUIZHI 2 Stück ESP8266 NodeMCU ESP12E WiFi Entwicklungsboard mit CH340 NodeMcu Lua WiFi 2.4 GHz V3 Modul kompatibel mit Arduino IDE/Micropython

Nein, nicht genau dieses Board, aber ein anderes mit ESP12F Modul. DAS ist das Entscheidende für WiFi, nicht die ganze Peripherie auf dem DevBoard.

Miklesoft, 2025-02-02:
Hallo, ich hatte auch die Problemem mit dem  8266 -12F. Habe in Platfomio.ini folgendes angehängt:
build_flags = 
	-D DEBUG_ESP_PORT=Serial 
  	-D DEBUG_ESP_CORE
  	-D DEBUG_ESP_HTTP_CLIENT
	-D DEBUG_ESP_WIFI
	-Os
Seit dem funktinoiert es einwandfrei.
Gruß
Michal

fs-sw, 2025-02-03:
> Hallo, ich hatte auch die Problemem mit dem 8266 -12F. Habe in Platfomio.ini folgendes angehängt: build_flags = -D DEBUG_ESP_PORT=Serial -D DEBUG_ESP_CORE -D DEBUG_ESP_HTTP_CLIENT -D DEBUG_ESP_WIFI -Os Seit dem funktinoiert es einwandfrei. Gruß Michal

Hi, in welcher Datei und an welcher Stelle hast Du das eingefügt.
Sorry, bin da Anfänger.
Danke

fs-sw, 2025-02-03:
> Hast du mal ausprobiert ob es auch geht wenn du nur -Os nutzt? Aus meiner Sicht ist das der einzige Flag der wirklich Einfluss haben sollte. Ist es wirklich nötig diverse Debugoptionen zu aktivieren um WiFi zum laufen zu bringen ist da irgendwas anderes im Argen - und so funktioniert es zufällig. Allerdings sollte -Os auch der Default sein wenn nichts angegeben ist.

Könnte das mein Problem sein?

Compiling .pio/build/nodemcuv2/lib973/DNSServer/DNSServer.cpp.o
.pio/libdeps/nodemcuv2/WebSockets/src/WebSocketsServer.cpp: In member function 'void WebSocketsServer::handleNewClients()':
.pio/libdeps/nodemcuv2/WebSockets/src/WebSocketsServer.cpp:649:96: warning: 'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]
  649 |         WEBSOCKETS_NETWORK_CLASS * tcpClient = new WEBSOCKETS_NETWORK_CLASS(_server->available());
      |                                                                                                ^
In file included from /Users/frankschleheck/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:40,
                 from .pio/libdeps/nodemcuv2/WebSockets/src/WebSockets.h:172,
                 from .pio/libdeps/nodemcuv2/WebSockets/src/WebSocketsServer.cpp:25:
/Users/frankschleheck/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiServer.h:85:14: note: declared here
   85 |   WiFiClient available(uint8_t* status = NULL) __attribute__((deprecated("Renamed to accept().")));
      |              ^~~~~~~~~
Compiling .pio/build/nodemcuv2/lib791/ESP8266WebServer/detail/mimetable.cpp.o
.pio/libdeps/nodemcuv2/PubSubClient/src/PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio/libdeps/nodemcuv2/PubSubClient/src/PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);

Vrumfondel, 2025-02-03:
Nein. Das ist normal.

fs-sw, 2025-02-03:
Danke

dbambus, 2025-02-21:
Hey @fs-sw 

konnte das Problem denn schon gelöst werden ?

Mit freundlichen Grüßen
David

fs-sw, 2025-02-22:
Leider nein.