ESPWortuhr/Multilayout-ESP-Wordclock #519: Weiss geht nicht mit übergängen, Symbole

JohannesWaltherGithub, 2025-01-26:
Hallo erstmal

zu beginn ein Kompliment an alles was da schon läuft.
Super Sache!

Ich habe noch ein Problem mit dem Weiss und den übergängen. Wenn ich Weiss auswähle (LED SK6812 RGBWW) wird nur weiss angezeigt solange kein Übergang eingestellt ist. Sobald ich die Art vom Übergang ändere, stellt die Farbe um (meist auf ein hellblau) und ich kann kein Weiss mehr anzeigen.
Die 3 Weisstypen (KW  NW  WW) habe ich alle bereits ausprobiert, macht aber kein unterschied.

Weiss nicht ob das ein falsche einstellung ist bei mir, oder ob es aus sonst einem Grund nicht geht.

Dann habe ich noch nicht herausgefunden wie man verschiedene Symbole anzeigen kann. Ich kann das Herz anzeigen und variabel schnell „blinken“ lassen. Aber wie ich zu allen andern Symbole komme (Brief, Glocke, Feuerwerk etc.)
Am liebsten auch gleich zum aktivieren per MQTT. Habe die Uhr auch damit in HomeAssistant eingebunden.

Wäre da dankbar um ein paar Tipps

Lg Joe

Vrumfondel, 2025-02-03:
Hallo!

Du hast völlig recht. Übergänge funktionieren aktuell nur mit RGB, nicht aber mit zusätzlichen weißen LEDs.
Was du da als "hellblau" siehst ist der Versuch mit den SK6812 RGB chips ein weiß darzustellen. Vermutlich gibt es für den Type einfach noch keine weißkalibration.

Ein Überblenden wie du es gerne hättest wäre ein neues Feature für den Sonderfall das ein Überblenden bei "weiß" mit Verwendung von RGBW LEDs erfolgen soll.

dbambus, 2025-02-03:
Hallo, 

die bescheidenen Transitionen, die sind damals als Paket gekommen und ich verstehe den Code noch nicht ganz und muss mich jedes Mal neu einarbeiten. 

Aber kurz gesagt. In den Transitions werden nur die RGB-Werte verwendet (gecached und editiert), das W kurzfristig hinzuzufügen klingt in der Theorie einfach, aber da müsste man wahrscheinlich ein paar Tests machen, dass das perfekt funktioniert. 

ABER: Ich denke es macht Sinn hier die komplette interne Verarbeitung der "LED Clock" von rein binär ( on/off für die Buchstaben in der LED.hpp) und RGB in den Transitions zu überarbeiten und eine neue Klasse zu erstellen, die die RGBW Werte für jede LED in der Matrix enthält. 

Das wird dauern xD zumindest bei mir. 

Mit freundlichen Grüßen
David

Vrumfondel, 2025-02-03:
... oder eine Funktion die generell RGB in RGBW Werte umrechnet ...

dbambus, 2025-02-03:
Hey, 

das ist schon von mir implementiert, zumindest für die Ausgabe auf die Matrix. Aber die Übergänge haben, vereinfacht gesprochen,  neben dem `set` auch ein `get` Befehl und dieser müsste angepasst werden, eben so das zwischenspeichern der Werte. deswegen wäre eine ganzheitliche Änderung wünschenswert, als einzelne Lösungen. 

Das wäre zumindest mein Gedanke :-) 

Mit freundlichen Grüßen
David

JohannesWaltherGithub, 2025-02-04:
Hey!
Vielen Dank für die Erklärungen. Ja das kann ich mir vorstellen, dass das Aufwändig ist. Mit programmiertechnisch leider nich so sattelfest um es selber zu machen. Ist aber auch halb so schlimm. Die Uhr läuft nun ohne Übergänge wunderbar.

Wie ist es mit den Symbolen? Hat das auch mit der Farbe zu tun? Oder kann dass an der verwendeten Sprache liegen (bei mir Schweizerdeutsch)?

Lieber Gruss

dbambus, 2025-02-21:
Hey @JohannesWaltherGithub, 

Die Symbole sind korrekt im Code vorhanden, aber noch nicht al Auswahl implementiert. Da habe ich schon paar Ideen, die werde ich aber erst zu einem späteren Zeitpunkt implementieren können. Ich will erst Ordnung bei den ganzen Issues schaffen. 

ToDo: 
Weitere Symbole. 

Mit freundlichen Grüßen
David