ESPWortuhr/Multilayout-ESP-Wordclock #521: Like to share my 3D Print and Step files like discussed

Korny3005, 2025-02-05:
Hey Mate,
ich würde gerne wie schon besprochen meine 3D Files zum Drucken teilen.
Ich hab alles zusammengepackt.

Kannst du die downloaden und in dein Projekt einfügen und zur Verfügung stellen?

https://1drv.ms/f/c/8826c8a57ae17475/EpGSsDImeghDheXVenSIZmIBRicLFrQk_2oaXXBH5pbB6Q?e=4e1tQm
PW: shadowcat

dbambus, 2025-02-05:
Hey @Korny3005, 

vielen Dank für die Daten, ich schaue es mir am Wochenende genauer an. Ich denke schon, dass ich das dem Projekt hinzufügen werde :-) 

Mit freundlichen Grüßen
David

Korny3005, 2025-02-05:
Ich glaub das hilft der Verbreitung ganz gut.
Weil weniger laser schneiden oder fräsen als 3D Drucken :)

merlin2533, 2025-03-16:
@dbambus  falls gewünscht habe Fräsdaten für die Front und auch für den Körper, das man es sehr leicht aufbauen kann.
Wenn gewünscht kann ich diese gerne für die 50cm Variante bereitstellen.

merlin2533, 2025-03-19:
Biete die gefräste Grundplatte auch an

https://www.etsy.com/de/listing/1391194420/word-clock-buchstabenwort-uhr?ref=shop_home_feat_2&frs=1&logging_key=9385fae6c11e592d5decd09a1efc408352030391%3A1391194420