ESPWortuhr/Multilayout-ESP-Wordclock #522: Minutes LEDs not active - not even on startup check

Korny3005, 2025-02-05:
Ich habe nur noch ein Problem.
Ich bekomme die Minuten LEDS nicht zum laufen.

Als wären die nicht da.
Kann es sein, dass die im Code in der verwendeten Variante nicht angesteuert werden?

Auch beim Startup, beim durchlaufen der LEDs leuchten diese nicht.

Wäre nett wenn du dir das anschauen könntest.

[image]

dbambus, 2025-02-05:
Hey, 

bei durchlaufen sollen sie nicht Leuchten (das wäre ein Feature Request), da soll nur die Matrix leuchten. Kannst du dich vergewissern, dass die korrekt angeschlossen sind. als 110 bis 114te LED ? 

Mit freundlichen Grüßen
David

Korny3005, 2025-02-05:
In reihe nach den anderen ja. 111.112.113.114.
Gibts da noch was zu beachten außer anschließen und auswählen?

closed 2025-02-21 by dbambus