ESPWortuhr/Multilayout-ESP-Wordclock #524: Horizontal / Vertical Mirroring funktioniert kurzzeitig nicht beim Bildaufbau

fex42, 2025-02-05:
Da ich bei meiner Uhr das Datensignal unten rechts statt oben links einspeise, musste ich horizontales und vertikales Spiegeln aktivieren. Leider treten hierbei Störungen auf: das Bild ist kurzzeitig immer mal ungespiegelt, wird dann aber sofort gedreht. Im ersten Moment dachte ich an Störungen aufgrund des längeren Datenkabels vom ESP8266, da aber das Bild einfach ungespiegelt erscheint und auch die Minuten-LEDs nicht betroffen sind, bin ich ziemlich sicher, dass dies ein Software-Bug und kein Hardware-Problem ist.

Hier ein Video, wo man das sehen kann. Die Störungen treten immer dann auf, wenn das Bild neu aufgebaut wird, in dem Fall durch Helligkeitsänderungen (LDR ist verbaut und aktiviert), wenn ich mich bewege.

https://www.youtube.com/shorts/xt7rbR0UdzI

(wenn der YT-Link unerwünscht ist, kann ich das Video auch hier direkt hochladen)

dbambus, 2025-02-05:
Hey @fex42, 

tritt der Fehler denn auch auf wenn eine Manuelle Helligkeitskontolle (ohne LDR) aktiviert ist ? 

Mit freundlichen Grüßen
David

fex42, 2025-02-05:
Gute Frage! Da die Uhr nicht mehr bei mir steht, kann ich das jetzt nicht austesten. Das hole ich aber spätestens am Wochenende nach. Ich meine aber den Effekt erst mit der Aktivierung des LDR im Menü gesehen zu haben (allerdings hab ich den ziemlich schnell aktiviert, daher bin ich nicht sicher).

Viele Grüße
Felix

fex42, 2025-02-06:
Hallo David,

es ist tatsächlich so, dass der Fehler nur dann auftritt, wenn man die automatische Helligkeitsregelung aktiviert. Mit manueller Helligkeit zeigt sich kein Bild-Kippen.

Viele Grüße
Felix

fex42, 2025-02-08:
Mir ist aufgefallen, dass teilweise weniger Argumente übergeben werden als manche Funktionen erwarten. Konkret ist dies beim Aufruf von `led.set()` in `ClockWork::loopAutoBrightLogic()` und dann der Aufruf von `setbyFrontMatrix(Foreground)` in der `Led::set(WordclockChanges changed)` Funktion, dem der das zweites Argument `applyMirrorAndReverse` fehlt.

Meine C++ Erfahrungen liegen sehr lange zurück und ich habe den Code (und das Tooling) auch nicht vollständig durchdrungen, aber vielleicht könnte das Ursache des Problems sein? Da ich die Uhr nicht bei mir stehen habe, kann ich das leider gerade schlecht testen, aber vielleicht kannst Du, @dbambus, bewerten oder mir einen Hinweis geben?

dbambus, 2025-02-21:
Hey @fex42, 

es sollte ggf. mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/508/commits/bc9a495ba99fcda496ab6ac158344147c542dc3a behoben worden sein :-) du kannst gerne mal diesen Commit testen und Feedback geben. 

Mit freundlichen Grüßen
David

closed 2025-02-21 by dbambus

fex42, 2025-02-21:
Hi @dbambus!

Ich geb auf jeden Fall Feedback, vielen Dank für den Fix! Wird ein paar Tage dauern, die Uhr steht ja nicht bei mir, sonst hätte ich es gleich gemacht. Schaffe ich hoffentlich am WE und geb dann Bescheid!

Übrigens meinte meine Frau, dass unsere Uhr auch ab und zu einen Fehler anzeigt (bei dieser Uhr hab ich die Standard-Verdrahtung, so dass kein Mirror aktiviert werden muss). Einmal hab ich den Fehler auch schon gesehen, war aber nur ganz kurz und kommt so selten vor, dass es nicht stört.

Viele Grüße
Felix