ESPWortuhr/Multilayout-ESP-Wordclock #530: Language issue

pienmatla, 2025-02-20:
Hi,

When i'ts 9:40, there are 2 options: ''twenty to ten'' and ''ten after half past nine''. 
In Dutch we say ''ten after half past nine''.

The option to switch between those settings doesnt work at my clock. It keeps saying twenty to ten. 

Can anyone help me with this issue?

Kind regards,
Pien

dbambus, 2025-02-20:
Hey @pienmatla, 

it is already fixed with https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/508/commits/e35abab16a6e960870409d4681080f34828d81e9

Cheers
David

closed 2025-02-20 by dbambus

pienmatla, 2025-02-21:
Hey @dbambus, at mine it still doesnt work in Dutch. Also the ''in words'' function for the minutes wont respond with the newest release.

Kind regards,
Pien