ESPWortuhr/Multilayout-ESP-Wordclock #542: Bei Dimmen via Sensor wird manchmal Rotation ignoriert

TychoGold, 2025-03-11:
Guten Abend,

vielen Dank für dieses wirklich schöne Projekt!

Ich habe die Uhr mit dem I2C Helligkeitssensor aufgebaut. Die LEDs sind horizontal gespiegelt eingestellt. Teilweise passiert es beim Dimmen, dass die Rotation für mehrere Sekunden verloren geht.

Viele Grüße
Tycho

dbambus, 2025-03-13:
Hey @TychoGold, 

vielen Dank für den Hinweis. Verwendest du die aktuellste Version aus dem Git oder das Release 4.0.0 ? Eigentlich sollte das Problem mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/commit/bc9a495ba99fcda496ab6ac158344147c542dc3a behoben worden sein. 

Mit freundlichen Grüßen
David

dbambus, 2025-03-14:
Ah okay, 

dann sollte das schon gefixed sein :-) 

Mit freundlichen Grüßen
David

closed 2025-03-14 by dbambus

TychoGold, 2025-04-07:
Guten Morgen,

[image]

Leider muss der Issue wieder aufgemacht werden. Ich hab die aktuelle Version auf drei Uhren und die Besitzer der zwei anderen Uhren berichten das gleiche.

Vielleicht liegt es an der Form der Rotation? Wir haben die horizontal gespiegelt aufgebaut.

Viele Grüße
Tycho