ESPWortuhr/Multilayout-ESP-Wordclock #547: ESP 32

merlin2533, 2025-03-15:
Hallo

habe jetzt über mehrere ESP 32 versucht in unterschiedlichen Modellen. Leider erhalten wir immer ein Problem mit dem Partitionen.
Hast du eine IDee?

SP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
E (234) esp_core_dump_flash: No core dump partition found!

Neuste Release 4.1 aber auch Release mit 3 versucht.
Danke

dbambus, 2025-03-15:
Hey @merlin2533, 

Ichhabe hier paar fragen. du nutzt den ESP32C3, wenn ich nur den DUMP hier anschaue. hast du das flashen über Plattformio probiert oder über die bereitgestellte .bin? 

Wenn Plattformio, hast du da die richtige build env in der plattformio.ini definiert? 

Ich denke bei der .bin kann es noch zu Fehlern kommen, da ich für den ESP32 das Partitionsschema angepasst habe und ich da erst noch den richtigen Befehl zur verfügung stellen muss. 

Mit freundlichen Grüßen
David

merlin2533, 2025-03-15:
Hallo David,

habe es mit Plattformio geflashed aber auch der BIN.
Env habe ich das richtige genommen. Hatte auch andere ESp32 bereits probiert die ich da hatte und versucht 
Leider beides ohne erfolg. mit der BIN aber auch direkt Plattformio.

Hast du mir einen TIP wegen Partitionen?
Sollte ja eigentlich mit Plattformio so wie ich den Code sehe auch richtig geflashed werden

Markus

sogidaned, 2025-03-16:
@merlin2533,

ich hatte eine ähnliche Fehlermeldung. hat dann immer neu gestartet. bei mir startet sie aber auch ohne esp_core_dump_flash.
mein Problem war die Initialisierung des I2C.  
Da ist im Programm 
#define SDA_PIN_ESP32 7
#define SCL_PIN_ESP32 6

mein ESP32dev board hat sie aber auf 21 und 22. sobald ich sie ändere startet die Software
#define SDA_PIN_ESP32 21
#define SCL_PIN_ESP32 22

Wegen den Partitionen da habe ich gerade diese laufen. Weil in dem aktuellem release auch kein OTA funktioniert hat beim ESP32
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x150000,
ota_0,    app,  ota_0,   0x160000,0x150000,
ota_data, data, ota,     0x2b0000,0x2000,
coredump, data, coredump,0x2b2000,0x8000,

grüße 
Alex

merlin2533, 2025-03-16:
Danke dir, habe gearde verschiedene Varianten versucht. Leider ohne Erfolg

Pin:: 21/22

mit ESp32C2
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
E (234) esp_core_dump_flash: No core dump partition fo⸮⸮⸮!⸮⸮⸮1!%⸮!1⸮g⸮1⸮1⸮⸮H⸮1!g⸮1!g⸮1!g⸮1!g⸮1!g⸮1!g⸮1!!!155⸮!1⸮⸮ !1D!!%⸮!%⸮⸮!!⸮!1D!⸮!⸮!1D⸮!1⸮1!!⸮!1⸮z!!⸮%⸮⸮⸮@5⸮⸮⸮@!%⸮⸮⸮1⸮!%⸮!1%!⸮$!⸮0⸮!⸮⸮⸮⸮!⸮⸮!%⸮!%!%⸮5!⸮ %⸮%!⸮ԕ⸮⸮ԡ%⸮⸮ԡ⸮⸮⸮ԩ(1%⸮1%%⸮⸮⸮4!!%⸮@⸮55⸮5⸮⸮⸮7⸮⸮
⸮⸮+⸮⸮c⸮ԡ⸮⸮55⸮5⸮⸮⸮7⸮1
⸮⸮+⸮⸮5⸮ԡ⸮⸮!
515$0⸮⸮!Ĕ⸮⸮⸮⸮⸮g⸮!!%⸮⸮⸮⸮7@!⸮⸮1⸮%$⸮⸮⸮-

mit ESP32 Wroom 32 (alte Partition
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (558) esp_core_dump_flash: No ⸮K⸮⸮dump partition found!
E (558) esp_core_dump_flash: No core dump partition found!
T⸮⸮⸮jU⸮⸮⸮jRUUUUU⸮⸮⸮⸮⸮⸮⸮UrAe⸮⸮*⸮]

Partitionen:
Wenn ich das verwende habe ich ein Problem mit dem RTC
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710

mit E32 Woom 32 (mit E32dev)
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (558) esp_core_dump_flash: In⸮Kɕ⸮сsize of core dump image: -1467940832
⸮⸮⸮⸮ZUUU⸮jR]UUUUUU⸮⸮⸮⸮⸮UrAe⸮u*⸮]

sogidaned, 2025-03-16:
@merlin2533,

habe leider die esp32C2 und esp32C3 nicht zum testen da. 
hab nur mal kurz die KI gefragt. 
Sie meinte:
Ja, es kann definitiv zu einem Konflikt mit LED_PIN 3 kommen, insbesondere bei ESP32-C3 und ESP32-C2, da GPIO3 bei diesen Varianten häufig für die serielle Kommunikation (UART TX) verwendet wird.

evtl gibt es bei den boards da auch noch einen Konflikt.
ich kann aber auch auf dem Holzweg sein.
hier noch eine liste von der KI.
Empfehlungsliste für Pin-Definitionen
ESP32 (Standard/ESP32Dev)
I2C:
SDA: GPIO21
SCL: GPIO22
LED (WS2812/NeoPixel):
Empfohlener Pin: GPIO2
Alternative: GPIO15, GPIO16, GPIO17
ESP32-C3
I2C:
SDA: GPIO8 oder GPIO9
SCL: GPIO6 oder GPIO10
LED (WS2812/NeoPixel):
Empfohlener Pin: GPIO8
Alternative: GPIO7, GPIO6
ESP32-C2
I2C:
SDA: GPIO5 oder GPIO7
SCL: GPIO6 oder GPIO8
LED (WS2812/NeoPixel):
Empfohlener Pin: GPIO4
Alternative: GPIO5, GPIO0
Wemos D1 Mini ESP32 (ESP32-WROOM-32)
I2C:
SDA: GPIO21
SCL: GPIO22
LED (WS2812/NeoPixel):
Empfohlener Pin: GPIO2
Alternative: GPIO13, GPIO15
ESP8266 (NodeMCU/Wemos D1 Mini)
I2C:
SDA: GPIO4 (D2)
SCL: GPIO5 (D1)
LED (WS2812/NeoPixel):
Empfohlener Pin: GPIO2 (D4)
Alternative: GPIO3 (RX)
Begründung
Die I2C-Pins sind jeweils die Standard-Pins, die vom Hardware-I2C-Controller unterstützt werden
Die LED-Pins wurden aufgrund ihrer Kompatibilität mit den RMT- oder Timer-Peripherien gewählt, die für die präzise Zeitsteuerung der WS2812-LEDs erforderlich sind
Für einige Chips (ESP32-C3, ESP32-C2) gibt es überlappende Empfehlungen für I2C und LED - in diesen Fällen muss abgewogen werden, welche Funktion welchen Pin bekommt
Diese Pin-Zuweisungen bieten die beste Leistung und Zuverlässigkeit für die jeweiligen Boards.

merlin2533, 2025-03-17:
Leider den ganzen Sonntag versucht aber leider ohne erfolg:
KI konnte mir auch nicht weiterhelfen aber auch die "Informationen aus dem Internet"

Für den ESP32-C3-DevKitM-1 ist es sehr wahrscheinlich, dass:
SCL auf GPIO6 (Pin 9)
SDA auf GPIO10(Pin 11) oder GPIO8(pin 6)

Fehler bleibt bei
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710

#define LED_PIN 8 // Use direct pin number
#define SDA_PIN_ESP32 9 // Use direct pin number
#define SCL_PIN_ESP32 11 // Use direct pin number

auch
#define LED_PIN 6 // Use direct pin number
#define SDA_PIN_ESP32 9 // Use direct pin number
#define SCL_PIN_ESP32 11 // Use direct pin number

Am ESP selbst ist auch nichts angeschlossen

dbambus, 2025-03-17:
Hey @sogidaned und @merlin2533, 

schon mal sehr cool, dass hier nach einer Fehlerursache gesucht wird. Ich kann aktuell nur beisteuern (Kann erst heute am Abend wieder Testen), dass die Software problemlos mit der aktuellen default config auf einem ESP32C3 und ESP8266 gelaufen ist. 

ich würde noch fragen, wisst Ihr welche Speicherconfig eure MCUs haben?

Als ESP8266 kommt bei mir ein WEMOS LOLIN MINi D1, ESP32 ein WEMOS 32 D1 Mini (4MB) und als ESP32C3 ein SuperMini (4MB) zum Einsatz. 

Mit freundlichen Grüßen
David

sogidaned, 2025-03-17:
@dbambus 

ich habe den esp32dev mit 4mb. 

[image]

bei läuft auch alles nach dem ich den i2c auf 21/22 gesetzt habe (leds habe ich nicht getestet)
ich kann auch leider den fehler von merlin2533 nicht mehr reproduzieren. 
kam noch von der zeit als ich dachte es lag an der LED Init und nicht der RTC Init die kurz darauf folgt.
bei mir hat die bootschleife immer nach LED Init begonnen. 
mein esp hat auch teilwese probleme bei der bootschleife und der hohen Baudrate überhaupt was anzuzeigen. 
bei einer monitorspeed von 115200 klappt es aber. Dann kommt mit der aktuellen software (pin 7/6) folgendes:
Beim start direkt:
E (558) esp_core_dump_flash: No ␇斕�dump partition found!
E (558) esp_core_dump_flash: No core dump partition found!
(läuft aber ohne core dump partition auch)

am ende des setups der rst fehler in schleife( aber nur bei pin7/6)
........
eeprom schreiben
LED Init
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

@merlin2533 

basieren auf der neuen info "ESP32-C3-DevKitM-1" :

ich weis nicht ob du die config schon versucht hast das würde cursor für das board vorschlagen.
soweit ich weiß, wird auch immer die gpio nummer angegeben. Nie die pin nummer vom board selber obwohl "PIN" in der Definition steht.

#define LED_PIN 8 
#define SDA_PIN_ESP32 5
#define SCL_PIN_ESP32 4

grüße
Alex

merlin2533, 2025-03-17:
glaub bei mir ist es verhext, habe ja auch das ESP Woom 32 Board

#define LED_PIN 8
#define SDA_PIN_ESP32 5
#define SCL_PIN_ESP32 4

wenn ich es mit Partion wie beim orginal , un deinen Pins Flashe, erhalte ich den Fehler immer noch

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (558) esp_core_dump_flash: No ⸮K⸮⸮dump partition found!
E (558) esp_core_dump_flash: No core dump partition found!
JUUUU⸮⸮⸮j⸮⸮JUU⸮⸮⸮⸮⸮⸮⸮⸮⸮UrAe⸮]e⸮YeS⸮⸮ bt⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮z⸮⸮Nw⸮U⸮⸮fU]HU⸮ ⸮ՅUUE⸮U⸮⸮E⸮UlU⸮T	⸮V⸮r⸮^]B⸮⸮m⸮⸮u4
⸮T⸮!⸮Zu⸮[⸮⸮⸮[⸮l⸮⸮[⸮du⸮⸮⸮⸮⸮⸮ur⸮q+ur⸮⸮ur⸮QuR⸮]⸮⸮⸮Iѝ⸮⸮⸮TemUi⸮/⸮U⸮⸮⸮E-ǆ-j⸮⸮qv⸮⸮⸮⸮}A⸮⸮⸮ets Jul 29 2019 12:21:46

und beim "ESP32-C3-DevKitM-1" :
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710

Auch wenn ihr mehrmals boote auch mehr Power gebe, Fehler ist immer der gleiche.

sogidaned, 2025-03-17:
@merlin2533 

setz mal die monitor speed runter von 460800 auf 115200. dann kann ich z.b. dass bei mir lesen:
x�x�␀�xxx��␀␀x��x�␀x␀��␀␀�x�x␀xx␀�␀�x��x�x�����x␀�␀␀

in src/wohrtuhr.cpp zeile 149
und in dem fall(esp32dev) in der platform.ini zeile 38

#define LED_PIN 3

// Define the I2C pins for the ESP32
#define SDA_PIN_ESP32 21
#define SCL_PIN_ESP32 22

dann läufts bei mir der anmeldehotspot

evtl noch einen erase flash vorher

grüße 
Alex

merlin2533, 2025-03-17:
Arbeitet so perfekt, @dbambus willst du das in den Master übernehmen? Oder dies als Option in env übernehmen?

esp32dev hier hat es perfekt funktioniert

merlin2533, 2025-03-17:
Mega Support

closed 2025-03-17 by merlin2533