ESPWortuhr/Multilayout-ESP-Wordclock #548: " LED´s pro Buchstabe

Yrrah1960, 2025-03-16:
Hallo zusammen,

erstmal vielen Dank für die ständige Weiterentwicklung des Programmes.

Ist es möglich bei "Bauart" auch die Funktion "2 LED´s entsprechen einem Buchstaben" zu integrieren.

Habe bei meiner Wortuhr in 1 x 1 m wegen der besseren Ausleuchtung der Buchstaben 2 LED´s pro Buchstabe verbaut.

Liebe Grüße

umakemetilt, 2025-04-18:
Ähnlich bei mir. Ich bräuchte eine Variante mit 4 LED pro Buchstabe in meiner 90er Variante. 

Die 45er Varianten sind großartig - ich habe inzwischen 10 davon gebaut und im Freundes- und Bekanntenkreis einige damit glücklich machen können.

BastiMu, 2025-04-19:
Müsste eigentlich relativ einfach umsetzbar sein. Bitte mal testen:

https://github.com/BastiMu/Multilayout-ESP-Wordclock/tree/Add-BuildTypes-x-LEDs-per-letter

Nach dem Umstellen auf x LEDs pro Buchstabe muss wie Wordclock neu gestartet werden, damit die Änderungen wirksam sind.

umakemetilt, 2025-04-23:
Seltsamerweise lädt bei mir kurz eine Version, bei erstmaligem Aufruf der Webpage, die z.B. eine Konfiguration für "Rahmen" (vermutlich ambilight gemeint?) zulässt. Das Icon wird aber unmittelbar ausgeblendet. Die Option die Anzahl der LED zu setzen finde ich gar nicht. Weißt du woran das liegen könnte? Nutze einen ESP8266.

BastiMu, 2025-04-24:
> [Verfolgungsbild][https://github.com/notifications/beacon/AS3CLDVZWNFIC45R236WS4T23AH6NA5CNFSM6AAAAABZDTK6AKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVINOYYG.gif]

Das die Option am Anfang kurz drin ist, ist normal. Das liegt daran, das das per Default implementiert ist und dann im Nachhinein für die nicht passenden Uhren wieder entfernt wird.

Komisch, bist du sicher, dass du den Branch von mir kompiliert hast? Schau mal in die index.html und ob da ca. in Zeile 400 die Optionen doubleRes, TrippleRes und QuadRes sind.

umakemetilt, 2025-04-24:
Erstmal vielen Dank für deine rasche Rückmeldung!
Kann ursächlich hierfür sein, dass ich noch nicht alles verdrahtet habe? Aktuell habe ich die ersten 3 Zeilen verlötet. 11x3x4 LED quasi. Nutze WS2812 4er Blöcke aus CN, die rückseitig verdrahtet werden. Aufgrund meines custom Layouts und der damit in Verbindung stehenden Anpassung der DE11x11.v3.hpp wollte ich zunächst mal testen, ob mein bisheriger Ansatz überhaupt funktionstüchtig wäre. Ich bin absolut kein Coder, meine aber die Logiken halbwegs verstanden zu haben und habe entsprechende Ansprache der LED/Words gem. meinem Layout angepasst. Ansonsten habe ich in deiner Version nur den nodemcuv2 in der platformio.ini als default_envs hinterlegt (das ändert aber ja meinem Verständnis nach nichts, da nur gezielter bzw fehlerfrei aufgespielt wird und quasi die Ladezeit verringert wird, richtig?) und meine WiFi Daten in der config.h ergänzt.

Erscheint auch die Option, mehrere LED pro Buchstabe wählen zu können analog zur ambilight Funktion - nur bei höherer Anzahl an LED? Das würde das ja dann erklären 😄 .

Danke jedenfalls für den Support!!

BastiMu, 2025-04-24:
Nee, das mit den zu wenig LEDs ist egal, der overflow wird einfach abgeschnitten. Und die 4er Blöcke sollten auch funktionieren, hab selber schon oft die 8x8 verwendet.

Das hinterlegte default_env sollte nichts ändern, allerdings meinst du nodemcuv3, weil v2 ist nicht definiert.

Die Option 4 LEDs pro Buchstabe sollte unter der Einstellung für die Wordcloud zu finden sein, so wo man vorher auch "Nur jede zweite LED" auswählen konnte. (Anzeigeoptionen -> Bauart)

Prinzipiell sollten deine Änderungen kein Problem sein, aber falls du Probleme hast würde ich erstmal eine nach der anderen einpflegen, bis es (nicht mehr) funktioniert. Also erstmal nur main aufspielen, um zu schauen, ob der Aufbau passt.  Dann die 4 LEDs pro Buchstabe. Dann das WLAN. Dann die Layout Änderung. usw.

Kein Ding, ich finde das Projekt super und interessant, was man alles für kreative Wordclocks bauen kann. Du kannst hier ja gerne mal ein Bild reinschicken, wenn sie fertig ist :)

umakemetilt, 2025-04-25:
Es funktioniert jetzt soweit. Kann dir gar nicht sagen, was das Problem war. 

Allerdings erscheint bei der 4er Variante ein 2er Layout. Siehst du, woran das liegen könnte?

umakemetilt, 2025-04-26:
[image]

umakemetilt, 2025-04-26:
[image]

Hier noch ein fertiges kleineres Format. Die Front ist lackiertes und gelasertes dc01 in Rostoptik und über neodym magnete fixiert. Gehäuse ist gelasertes mdf. Gesamtstärke liegt bei 20mm inkl Front.

umakemetilt, 2025-05-02:
Die Anpassung auf 2 LED funktioniert ordnungsgemäß. Die Anpassung auf 3 oder gar 4 LED je Buchstabe führt auch bei Neustart nicht zu einer Änderung. Es bleibt dann bei 2 LED je Buchstabe. Ansonsten sind die Minuten nicht am Ende (Ecken) sondern mittig im Feld. 

[image]

Grüße!

AndiK66, 2025-05-04:
Hallo zusammen,
ich habe eine Uhr in 90x90cm gebaut. 2 LEDs pro Buschstabe. Alles an einem ESP8266. Läuft perfekt. Ist nur eine große Geduldsaufgabe die LEDs zu zerschneiden und alles wieder aneinanderlöten. 

[image]
[image]

Grüße!

umakemetilt, 2025-05-04:
Interessanter Ansatz. Du hast den Datenstrang quasi gesplittet und somit werden 2 LED Stränge parallel angesprochen. 
Der restliche Aufbau ist soweit sehr ähnlich zu meinem (bei den 45cm hab ich WS2812B genommen - da gibt es keinerlei Probleme mit den Daten und der Workaround ist nicht nötig nach dem ersten Buchstraben). 

Ich nutze allerdings für die 90er Variante fertige 4er Blöcke WS2812, da funktioniert das leider so in der Form nicht. Bei mir ist das Anpassen des Codes unumgänglich - nur leider reichen meine Skills dafür nicht. 

Es bestehen ja im Grunde 2 Probleme:
-die Minutenposition müsste in getminutearray über numpixelswordmatrix errechnet werden und dann entsprechend bei mir auf Position 485 - 500 liegen (11x11 Layout zzgl. 4 LED je Buchstabe => 121 * 4 = 484; Ab Pos 485 je 4 folgend pro Minute).
-aktuell kann ich durch die Erweiterung von BastiMu auf 2 LED je Buchstabe stellen. Es gilt für mich dann zunächst nachzuvollziehen, wie die Logik dahinter aufgebaut ist und wie ich das entsprechend angepasst bekomme. 

Einerseits korreliert das eine mit dem anderen, andererseits versuche ich mir aktuell ein Bild zu machen, wo die entsprechenden Anpassungen zusätzlich zu hinterlegen wären. Für mich alles Neuland und schwer lösbar. Ich hoffe auf Hilfe von BastiMu.

BastiMu, 2025-05-04:
@AndiK66 Coole Idee, ich wäre da wahrscheinlich beim Löten durchgedreht :D

@umakemetilt ich schaus mir später oder morgen nochmal mal an, bin erst aus dem Urlaub zurück gekommen ;)

BastiMu, 2025-05-05:
@umakemetilt So hab jetzt mal geupdated, versuch bitte nochmal, ob bei dir jetzt alles klappt.

https://github.com/BastiMu/Multilayout-ESP-Wordclock/tree/Add-BuildTypes-x-LEDs-per-letter

umakemetilt, 2025-05-05:
> [@umakemetilt](https://github.com/umakemetilt) So hab jetzt mal geupdated, versuch bitte nochmal, ob bei dir jetzt alles klappt.
> 
> https://github.com/BastiMu/Multilayout-ESP-Wordclock/tree/Add-BuildTypes-x-LEDs-per-letter

Erstmal vielen lieben Dank für Deine Mühen! Hätte nicht sofort sein müssen ;-)

Tatsächlich funktioniert jetzt die Darstellung mittels 4 LED je Buchstabe. Die Minuten liegen nur noch immer auf LED 117-120. Einzeln, also nicht in 4er Blöcken.

BastiMu, 2025-05-07:
Okay, was meinst du mit viele crashes? Im Prinzip könnte hier ein Laufzeit Problem auftreten (es wird jede Sekunde ein Update gecalled), da es ich glaube keine Wordclock mit so vielen LEDs gibt. Eventuell kannst du ja mal den Serial Monitor mitlaufen lassen um zu sehen, wo es abstürzt.

Hab die Minuten nochmal korrigiert, da war ein Fehler drin. Und auch beim ablöschen der Minuten. Schau bitte nochmal, ob's jetzt klappt.

umakemetilt, 2025-05-07:
> Okay, was meinst du mit viele crashes? Im Prinzip könnte hier ein Laufzeit Problem auftreten (es wird jede Sekunde ein Update gecalled), da es ich glaube keine Wordclock mit so vielen LEDs gibt. [...]

Hey! Danke dir für die Rückmeldung. Heute war alles seltsamerweise deutlich stabiler. Vermutlich einfach ein lokales Problem - ich beobachte das aber weiterhin. War bislang auch weitestgehend mit dem Tausch defekter Blöcke beschäftigt. Super hohe Ausfallquote leider. 

Die Minuten sind leider unverändert in der dritten Zeile. Alle Einstellungen (Ecken, 4 LED) hatte ich sicherheitshalber ebenso getestet. 

Fürth... Ich bin keine 30km entfernt - notfalls muss ich vlt doch ins Auto setzen :) 
[image]

umakemetilt, 2025-05-08:
-Die Minuten werden zwar in den 4er Blöcken angezeigt, allerdings konstant in hellem Grün im Vordergrund. Die Farbänderungen für Vordergrund werden also nicht übernommen. Hintergrund hingegen schon.
-Jetzt funktioniert alles bis zum 30. Block inkl. der Minuten, aber...
-Ab dem 31. Block geht allerdings nach jeglicher Einstellungsänderung (ich denke nach jedem Impuls, weil auch eine Veränderung der Uhrzeit den selben Effekt auslöst, bzw. da einmal ganz kurz alles aufleuchtet) unmittelbar alles aus. Heißt also : man ändert die Farbe, die Änderung wird übernommen. Nach etwa einer Zehntel Sekunde (reine Schätzung) gehen aber alle LED Blöcke nach dem 30. aus. Im 31. Block leuchtet nur eine einzige LED (in korrekter Farbgebung, wie alles andere auch). Nur die Minuten bleiben darüber hinaus beleuchtet (Hintergrund korrekt, Vordergrund hellgrün unabhängig von der Farbwahl für Vordergrund). 
-Ändert man die Hintergrundhelligkeit (ich nutze ein 5V 10A Netzteil und müsste demnach eigentlich alle LED versorgen können), verringert sich im unteren Drittel die Helligkeit des Vordergrundes. 

Ich bezweifle, dass das irgendetwas ändert, aber denkbar wäre in meiner Version sicherlich eine LED je Minute. Ich habe in meiner Frontplatte Öffnungen mit 2mm Durchmesser (die Blöcke sind also eh überdimensioniert). Nach reset des ESP und neuem Aufspielen der Software läuft übrigens tatsächlich alles flüssiger und der ESP ist dauerhaft erreichbar. Danke dir abermals! Echt super, dass du dir die Mühe machst! Ich hoffe sehr, dass es eine Lösung für mein Problem gibt.

umakemetilt, 2025-05-18:
[image][image][image]

Aktueller Stand ist schon recht schön. Wenn jetzt noch die Uhr ginge 🤤

BastiMu, 2025-05-21:
Sorry, bei mir ist gerade etwas viel los..

Bezüglich der Helligkeit: Eventuell musst du öfter die 5V einspeisen, da bei langen Leitungen die Spannung sinkt. (Also z.B. bei jeder 2. Reihe die 5V und GND kappen und neu mit dem Netzteil verbinden.

Schwierig so aus der ferne zu Debuggen. Hast du nächsten Mittwoch Abend Zeit? Dann könntest du ja mal im FabLab vorbeischauen, dann können wir da mal gemeinsam draufschauen?

umakemetilt, 2025-05-28:
Erstmal vielen Dank für den input. Tatsächlich war das auch scheinbar das Problem. Jetzt funktioniert soweit alles uneingeschränkt. Ich hatte bislang nur ein paar "Probeläufe" gemacht und die Uhr noch nicht aufgehängt. Die letzte, finale Problemstellung wäre lediglich folgendes:

-zur vollen Stunde hin wird im 11x11 Layout scheinbar immer das Wort "Uhr" mit beleuchtet. Der Fehler tritt ab "viertel vor X" auf. Siehe Fotos. Das "Uhr" sollte ja eigentlich nur bei vollen Stunden angesprochen werden.

-"Es ist Ein Uhr" ist korrekt im Code hinterlegt. Also sowohl "ein" als auch "eins". Es wird dennoch "es ist eins Uhr" angesprochen. 

Kannst du das nachvollziehen? Danke dir für die Hilfe! Ich werde auf jeden Fall mal bei euch vorbeischauen. Heute schaff ich es leider nicht. 

Hier ein Beispielbild vom "Fehler".

[image]

umakemetilt, 2025-05-28:
hab die Datei mal hochgeladen: https://uploadnow.io/f/mrVkzxT

BastiMu, 2025-05-31:
Der Fehler mit UHR ist schnell gefunden: Nach `FrontWord::vor:` fehlt ein break (ist im original code auch schon falsch)

```
        case FrontWord::vor:
        case FrontWord::v_vor:
            setFrontMatrixWord(3, 8, 10);
            break;

        case FrontWord::uhr:
            setFrontMatrixWord(10, 0, 2);
            break;
```

BastiMu, 2025-05-31:
Bezüglich der "Es ist" Einstellung: Ich hab's bei mir getestet und bei meiner Uhr wird es gespeichert. Auch im Code wird korrekt `eeprom::write();` () nach dem Ändern der Einstellung ausgeführt. Nur nach einem neuen Upload wird eben die Einstellung überschrieben, aber nach einem Neustart müsste die Einstellung gespeichert bleiben.

Oder wird bei dir die Einstellung korrekt angezeigt, "Es ist" aber permanent angezeigt?

umakemetilt, 2025-06-01:
Super. Hab das im Code angepaßt und es läuft alles wunschgemäß.

Vielen lieben Dank dir nochmal für deine Hilfe! Ist absolut nicht selbstverständlich!

BastiMu, 2025-06-03:
Kein Ding, freut mich, dass jetzt alles klappt :)

Da jetzt alles funktioniert mach ich mal einen Pull Request, das Problem mit dem "Es ist" würde ich mal auf einen neuen issue verschieben, ich glaube das sind schon genug Änderungen für einen PR. Danke fürs testen.