ESPWortuhr/Multilayout-ESP-Wordclock #550: [bug] Animationen zerstören Wort Anzeige (V4.1)

udo1toni, 2025-03-21:
Ich habe gerade das Update auf 4.1 eingespielt. 
Wenn ich jetzt einen Übergang aktiviere (schieben, Matrix, whatever...), werden statt der korrekten Zeit in Worten (das ist der ausgewählte Modus) die beiden letzten Stellen der IP angezeigt - um ein Pixel nach links verschoben, d.h. es fehlt auch noch die erste Spalte der ersten Ziffer.
Die Anzeige ist wieder normal, sobald der Übergang abgeschaltet und einmalig ein Refresh ausgelöst wurde (also z.B. mittels Umschalten auf einen anderen Modus und wieder zurück zur Wortuhr)

Ich nutze das Standard Layout (äh... 10 x 11?) und einen GRB Strip, falls das relevant sein sollte...

Da ich die Übergänge nicht nutze, ist das für mich nicht wirklich relevant :) aber trotzdem...

Die aktualisierte Weboberfläche gefällt mir sehr 👍

D-ominik, 2025-04-03:
Ja, die Version ist leider sehr buggy. Kann dieses Problem bestätigen, Übergang ist unbrauchbar.

Weiterhin führt bei mir die Anzeige "Sekunden" sowie "Digitaluhr" zum sofortigen Reboot.

Und noch schlimmer: Zurücksetzen auf Standardwerte killt die Uhr komplett, danach muss der Flash erased und neu geflasht werden, sonst geht gar nichts mehr...

dbambus, 2025-04-04:
Hey @D-ominik und @udo1toni,

Ich hab hier erstmal zwei Fragen. Wenn man die Uhr ohne Animationen betreibt, treten dann auch die Bugs zum Vorschein? Welche MCUs werden von euch verwendet?

Das hilft mir mögliche Fehler einzugrenzen. 

Mit freundlichen Grüßen
David

D-ominik, 2025-04-04:
Hi @dbambus,

in meinem Fall ist es ein NodeMCU Lolin V3 mit SK6812 RGBW.

Der Reboot bei Aktivierung der Sekunden oder Digitaluhr Anzeige ist unabhängig vom Übergangseffekt. Ebenso beim Reset auf Standardwerte.

udo1toni, 2025-04-06:
Hallo @dbambus bei mir ist es ein D1 Mini (wenn ich mich nicht maßlos täusche...)

hoffie, 2025-04-06:
> Weiterhin führt bei mir die Anzeige "Sekunden" sowie "Digitaluhr" zum sofortigen Reboot.
>
> Und noch schlimmer: Zurücksetzen auf Standardwerte killt die Uhr komplett, danach muss der Flash erased und neu geflasht werden, sonst geht gar nichts mehr...

@D-ominik Kannst du mal testen, ob #558 diese Probleme für dich fixt?
(Dort ist zwar auch ein Fix in der Transition-Logik enthalten, soweit ich das sehe, betrifft das aber nur den Countdown, d.h. mit den anderen hier genannten Transition-Problemen hat das vermutlich nichts zu tun)

hoffie, 2025-04-06:
> Ich habe gerade das Update auf 4.1 eingespielt.
> Wenn ich jetzt einen Übergang aktiviere (schieben, Matrix, whatever...), werden statt der korrekten Zeit in Worten (das ist der ausgewählte Modus) die [...]

Den Fehler kann ich auch nachvollziehen, allerdings wirklich nur direkt nach dem Starten der Uhr und bis zum ersten Wortwechsel. Ab dann passt es, soweit ich es sehe.

Unten stehender Patch fixt es für mich. Ich teste ihn aber noch ein paar Tage, bevor ich ihn als PR einreiche. Nicht, dass er etwas anderes kaputt macht.

@udo1toni @D-ominik: Tritt der Fehler bei Euch noch in anderen Fällen bzw. länger auf? Habt ihr Hinweise darauf, wann das zuletzt korrekt funktionierte? Besteht die Chance, dass das schon länger so war und lediglich jetzt bei genauerem Blick direkt nach Uhrstart aufgefallen ist? Ich sehe auf dem Weg zu 4.1.0 auf Anhieb keine Änderungen, die das auslösen könnte. Wenn ich mich recht erinnere, habe ich durch Wechseln der Effekte schon in früheren Versionen seltsame Zustände ("hängende" Pixel, falsche Farben) erzeugen können, die sich erst mit dem nächsten Wortwechsel korrigiert haben.

```
diff --git a/include/clockWork.hpp b/include/clockWork.hpp
index 3d33ebc..5284f2b 100644
--- a/include/clockWork.hpp
+++ b/include/clockWork.hpp
@@ -1100,6 +1100,9 @@ void ClockWork::clearClockByProgInit() {
         G.progInit = false;
         led.clear();
         led.show();
+        // this discards the previous frame and avoids basing future
+        // transitions on old state:
+        transition->init();
     }
 }
 
```

D-ominik, 2025-04-09:
@hoffie Der Reboot ist mit #558 behoben, die Sekundenanzeige funktioniert auch. Die Digitalanzeige hat jedoch noch einen Bug:
normalerweise steht in der oben Hälfte die Stunde, in der unteren Hälfte die Minute. Mit dem Update verschwindet eines von beidem im Zwei-Sekunden-Takt, d.h. es wird entweder die Stunde oder die Minute angezeigt.

Seit wann das Problem besteht, kann ich nicht sagen, weil ich meine Uhren immer noch mit 3.3 betreibe. Die im letzten September oder diesen Monat gebauten und verschenkten Uhren bekamen jeweils die aktuelle Firmware, um dann nach kurzen Tests wegen Bugs auch eine 3.3er Firmware zu erhalten. Möglicherweise bestand das Problem schon bei #434.

hoffie, 2025-04-09:
@D-ominik:
> Die Digitalanzeige hat jedoch noch einen Bug: normalerweise steht in der oben Hälfte die Stunde, in der unteren Hälfte die Minute. Mit dem Update verschwindet eines von beidem im Zwei-Sekunden-Takt, d.h. es wird entweder die Stunde oder die Minute angezeigt.

Ist mir auch aufgefallen, ist aber anscheinend nicht grundsätzlich ein Bug, sondern ein Feature für "kleine Layouts". Aus meiner Sicht würde ich das 10x11-Layout nicht als "klein" bewerten und somit gerne das 3.3er-Verhalten beibehalten.

Die Verhaltensänderung kam durch diese Code-Änderung, wo ich auch schon @BastiMu und @dbambus angepingt habe, um zu klären, ob das wirklich so beabsichtigt war:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/483/files#r2030287081

Folgender Patch würde da wieder das alte Verhalten für 10x11-Uhren herstellen:
```diff
diff --git a/include/led.hpp b/include/led.hpp
index 1d31b3e..c854763 100644
--- a/include/led.hpp
+++ b/include/led.hpp
@@ -710,7 +710,7 @@ void Led::showDigitalClock(const char min1, const char min0, const char h1,
     bool showHours = true;
     bool showMinutes = true;
     // toogle hours and minutes if clock is not high enough
-    if (usedUhrType->rowsWordMatrix() < (pgm_read_byte(&(fontHeight[usedFontSize])) * 2 + 1)) {
+    if (usedUhrType->rowsWordMatrix() < (pgm_read_byte(&(fontHeight[usedFontSize])) * 2)) {
         if (_second % 4 < 2) { // show hours every 2 seconds
             showHours = true;
             showMinutes = false;
```

D-ominik, 2025-04-09:
> Ist mir auch aufgefallen, ist aber anscheinend nicht grundsätzlich ein Bug, sondern ein Feature für "kleine Layouts". Aus meiner Sicht würde ich das 10x11-Layout nicht als "klein" bewerten und somit gerne das 3.3er-Verhalten beibehalten.
> 

Ah, ok. Diese Änderung macht irgendwie gar keinen Sinn:

- Wenn auf einem kleinen Layout ständig die Anzeige zwischen Stunde und Minute wechselt, ist das derartig hektisch, dass man das niemals praktisch so nutzen wird. Das ist ein klassischer Fall, der als auf-diesem-Gerät-nicht-verfügbar behandelt werden muss.
- Auf der 10x11 passt die Digitaluhr komplett drauf, war ja auch in vorigen Versionen so. Jetzt wird halt in der oberen Hälfte kurz die Stunde gezeigt (untere Hälfte leer), dann ausgeblendet, dann werden in der unteren Hälfte die Minuten sichtbar (obere Hälfte bleibt leer), dann kehrt sich das wieder um. Das ist in dieser Form völliger Nonsense und macht die Digitaluhr auch auf der 10x11 unbrauchbar.

D-ominik, 2025-04-09:
> Evtl. würde es aber etwas bringen `led.show();` vor `transition->init();` zu setzen.

Ändert am Verhalten leider nichts.

udo1toni, 2025-04-12:
Ich mache das Update vom Schreibtisch aus (1.  Stock), die Uhr hängt aber im Wohnzimmer (Erdgeschoss). Ich bin auch eher gemütlich unterwegs :) 
Was ich damit sagen will: Die Anzeige hat ziemlich sicher zwischendurch schon die angezeigten Worte gewechselt (beschwören kann ich das natürlich nicht).
Die Effekte habe ich eigentlich bei jeder neuen Version immer mal kurz angetestet, und ja, da gab es gerne mal Fehler (allerdings gewöhnlich nur beim Matrix Effekt)
Bei den letzten Versionen war mir nichts aufgefallen, sonst hätte ich einen Issue erstellt. 
Kann aber auch gut sein, dass ich einfach nicht zum Testen gekommen bin...

Im Normalbetrieb ist mir nichts weiter aufgefallen.

Eine grundsätzliche Sache (die ist aber komplett unabhängig und eher ein Schönheitsfehler...):
Wenn ich ein Update einspiele, behält die Uhr gewöhnlich die WLAN Konfiguration (prima!), die restlichen Einstellungen gehen aber verloren (insbesondere welche LEDs verbaut sind, ich habe hier GRB, nach dem Flashen kommt die Uhr aber immer mit BRG hoch. Mache ich da was falsch, oder ist das ein bekanntes Verhalten?

dbambus, 2025-04-13:
@D-ominik 

Ist noch immer der Fehler vom Zurücksetzen vorhanden?

D-ominik, 2025-04-13:
Jein, das war eine Fehlinterpretation. Die Anzeige bliebt komplett leer (warum auch immer) und das Captive Portal war nicht nutzbar (träge und mit Timeouts). Daher erschien es, als hätte der Reset den MC abgeschossen. Tatsächlich war das Captive Portal bei späterem Versuch mit gleichem Verhalten doch normal erreichbar, nachdem ich mein Telefon mal neu gestartet hatte.

hoffie, 2025-04-21:
> Ist mir auch aufgefallen, ist aber anscheinend nicht grundsätzlich ein Bug, sondern ein Feature für "kleine Layouts". Aus meiner Sicht würde ich das 10x11-Layout nicht als "klein" bewerten und somit [...]

Nachdem inzwischen [bestätigt](https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/483#discussion_r2030287081) wurde, dass diese Änderung unbeabsicht war, habe ich einen PR mit dem Fix submitted: https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/580