ESPWortuhr/Multilayout-ESP-Wordclock #552: Minuten LED Farbe Unterschiedliche

User-HO, 2025-03-28:
Ich habe das Problem erkannt, das wenn die automatische Helligkeit eingestellt ist, die LED Farben der Minuten (4Stk.) andere Farben anzeigt wenn sich die Helligkeit ändert.
Beispiel bei mir, Hintergrund Blau, Wort Farbe Pink.
Ändert sich die Helligkeit, ändern sich die Minuten Farben in Rot und halten die Farben dann auch bei, bis diese neu Beschrieben werden bzw. zugeordnet werden.

dbambus, 2025-03-29:
Hallo @User-HO, 

vielen Dank für die Meldung des problems, ich werde mir das anschauen. :-) 

Mit freundlichen Grüßen
David

hoffie, 2025-04-07:
Mir ist der Fehler auch gerade aufgefallen und ich kann ihn gezielt nachstellen:
- Ausgangsbedingungen: Vordergrund-Farbe z.B. orange, mind. eine leuchtenden Minuten-LED
- LDR-Sensor abdunkeln, wieder normal beleuchten, wieder abdunkel, wieder Licht ranlassen. Nach einigen Malen ist deutlich erkennbar, dass sich der Farbton langsam von orange auf rot ändert.

Es sieht so aus, als würde das Problem durch den Detailverlust bei Helligkeitsanpassungen passieren. Ich finde aber im Code keine konkreten Hinweise darauf.

Manchmal blinken die Minuten-LEDs auch im Sekundentakt in der richtigen Farbe auf, um dann aber die meiste Zeit wieder in der falschen Farbe zu leuchten.

Unabhängig von Helligkeitsänderungen (vielleicht sogar unabhängig vom Auto-LDR?) tritt ein weiterer, verwandter Fehler auf: Ändert man die Vordergrundfarbe, so bleiben die Minuten-LEDs bis zur nächsten Minute 0 in der alten, falschen Farbe.

f9a42862eb339d38233901d4f74a981beea20bdf scheint der Commit zu sein, der den Fehler auslöst. Reverted man ihn, funktioniert alles wieder korrekt (allerdings gab es ja vermutlich einen Grund für die damalige Änderung; vermutlich ist der Commit korrekt, aber evtl. wird foregroundMinute nicht zuverlässig geupdated, was dann nur durch diesen Commit zum Tragen kommt).

dbambus, 2025-06-10:
Hey, 

Ich habe mit dem aktuellen pre-Release 4.2.0 Änderungen an der Transition.hpp vorgenommen um das Problem gezielt zu adressieren. Könnt Ihr das nochmal testen ? 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V4.2.0

Mit freundlichen Grüßen
David

dbambus, 2026-10-23:
Hallo @User-HO, @hoffie und @umakemetilt,

vielen Dank für die Meldung und vor allem für die genaue Analyse. @hoffie, dein Hinweis, dass das Zurücklesen der Farben vom LED-Streifen fragil ist, war genau der richtige Riecher. Das habe ich in 5.0 abgeschafft: Farben und Übergänge laufen jetzt über eine eigene Render-Pipeline, und die Minuten-LEDs bekommen ihre Farbe bei jedem Neuzeichnen direkt aus der eingestellten Farbe inklusive Helligkeit. Damit sollte die Farbe weder bei Helligkeitswechseln durch den LDR wegdriften noch, @umakemetilt, bei aktiviertem Übergang nach einem Farbwechsel in der alten Farbe hängen bleiben.

Enthalten im Release Candidate V5.0.0-rc.1:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V5.0.0-rc.1

Achtung: Mit dem Update werden alle Einstellungen zurückgesetzt.

Ich schließe das Issue. Falls ihr das Verhalten mit der neuen Version noch seht, öffnet es gerne wieder.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus