ESPWortuhr/Multilayout-ESP-Wordclock #553: Layout bearbeiten

bytec77, 2025-03-29:
Hallo Zusammen. 

Erstmal Lob an das ganze Projekt hier. 

Ich habe es soweit auch umgesetzt nur passt bei mir das Layout nicht ganz wie ihr auf dem Bild sehen könnt. 

Wie mache ich das am besten? Eine bestehende .hpp Datei bearbeiten? 

Gruß

Chris

[image]

SkylightXD, 2025-03-29:
Das gleiche Problem habe ich auch. Meine erste LED ist unten Links, ich habe das Layout also gedreht und gespiegelt ...

bytec77, 2025-03-29:
Meine erste LED ist oben link. Es ist ja auch nur bis jetzt das Problem unten rechts MUH statt UHR :-)

dbambus, 2025-03-29:
Hey @SkylightXD @bytec77,

ich find Muh doch gut :-) @bytec77 Dein Layout entspricht nicht dem DE10x11.2Clock sondern dem DE10x11 Dann sollte das auch mir der Uhr klappen :-) 

@SkylightXD Drehen und Siegeln sollte in der Software funktionieren. eine Matrix Transponieren geht noch nicht, vielleicht ja bald. 

Mit freundlichen Grüßen
David

bytec77, 2025-03-29:
@dbambus 

Stimmt schlecht ist Muh auch nicht :-) 

aber im DE10x11 ist auch keine MUH :-) 

also passt das Layout leider auch nicht 

[image]

bytec77, 2025-03-29:
theoretisch müsste ich ja da anscheinend nur UHR falsch ist folgendes:

`case FrontWord::uhr:
            setFrontMatrixWord(9, 1, 3);   // "UHR"
            break;`

in: 

`case FrontWord::uhr:
            setFrontMatrixWord(9, 0, 2);   // "UHR"
            break;`

ändern oder ?

dbambus, 2025-03-29:
Hey, 

genau. Ah jetzt hab Ichs. Das Nero Layout passt 

[image]

closed 2025-03-29 by dbambus

SkylightXD, 2025-03-30:
Super ! Neo hat geholfen :)