ESPWortuhr/Multilayout-ESP-Wordclock #557: Crash in v4.1.0 when setting mode digital clock or seconds (potentially in Led::showNumbers())

hoffie, 2025-04-06:
#550 contains comments about crashes in v4.1.0. I'm experiencing the same. I'm opening this as a sperate issue as #550 is mainly about distorted animations and I don't want to conflate those topics.

## Trigger
The crash happens when setting the mode to digital clock or seconds. It might also trigger directly after boot (probably if digital clock had been selected before booting/flashing the new version).

## Exception

```
Exception (3):
epc1=0x4020bf0d epc2=0x00000000 epc3=0x00000000 excvaddr=0x402ac18e depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x4020bf0d in Led::showNumbers(char, char) at ??:?
  excvaddr=0x402ac18e in system_get_sdk_version at ??:?

>>>stack>>>

ctx: cont
sp: 3ffffd00 end: 3fffffd0 offset: 0150
3ffffe50:  e88ee888 e8eee8e8 3ffffe80 3ffffe70  
3ffffe60:  00000008 00000034 00000032 40205e41  
3ffffe70:  3fff22a2 3ffe8648 00000002 0000001a  
3ffffe80:  3fffff31 00000000 3fff51a4 3fff26f4  
3ffffe90:  3fff22a2 3ffe8648 3ffea9f4 4021449c  
3ffffea0:  35320032 2d34302d 31203630 36343a34  
3ffffeb0:  2032343a 32302b28 0a293030 3ffe8300  
3ffffec0:  3ffe8368 402b0810 3ffe8368 40236f52  
3ffffed0:  00000000 00005b0b fe353f7c 006945dd  
3ffffee0:  00000000 00000000 3ffe86f8 00000000  
3ffffef0:  00000000 00000000 00005b0b 40228885  
3fffff00:  00000000 00005b0b fdf3b645 006945dd  
3fffff10:  3fff2658 00000001 3fff7a40 000000cc  
3fffff20:  00000000 00005b0b 00000000 4022984c  
3fffff30:  00000034 3fff1a40 3fff1da0 40218088  
3fffff40:  4022f0a8 00000000 00001388 3fff28d8  
3fffff50:  3fff1e6c 00005b0b 00000000 4022984c  
3fffff60:  3fff2371 3fff236f 3fff1a40 3fff28d8  
3fffff70:  3fff2371 3fff236f 3fff18c8 402149d6  
3fffff80:  0000002a 0000002e 0000000e 00000006  
3fffff90:  00000003 0000007d 00000000 0000005f  
3fffffa0:  00000001 feefeffe 67f277b2 00000000  
3fffffb0:  3fffdad0 00000000 3fff28ac 40228930  
3fffffc0:  feefeffe feefeffe 3fffdab0 401003e1  
<<<stack<<<

0x40205e41 in _ZN23NeoEsp8266DmaMethodBaseI19NeoEsp8266DmaEncodeI25NeoEsp8266DmaSpeedWs2812xEE6UpdateEb$isra$0 at Wortuhr.cpp:?
0x4021449c in ClockWork::loop(tm&) at ??:?
0x402b0810 in etharp_output at ??:?
0x40236f52 in _getenv_r at /workdir/repo/newlib/newlib/libc/stdlib/getenv_r.c:126
0x40228885 in esp_try_delay(unsigned int, unsigned int, unsigned int) at ??:?
0x4022984c in __delay at ??:?
0x40218088 in WebSocketsServerCore::handleClientData() at ??:?
0x4022f0a8 in DhcpServer::isRunning() const at ??:?
0x4022984c in __delay at ??:?
0x402149d6 in loop at ??:?
0x40228930 in loop_wrapper() at core_esp8266_main.cpp:?
0x401003e1 in cont_wrapper at ??:?
```

## Hardware
Node MCU: D1 Mini Pro 4m
LED band: WS2812B Grb
Layout: Ger10x11Clock (my gut feeling is that the layout dimensions might be most relevant factor, but I haven't verified yet)

## Analyses / Potential causes
git bisect pointed me to 1ea211006d787bfdb86d4fc258f8a7bd153a149e as the first broken version (cc @BastiMu). I'm currently trying to further pinpoint the exact details.

dbambus, 2025-04-06:
Hey @hoffie,

Thank you for the detailed bug report. Can you elaborate if the last release is also affected ?

Cheers
David

hoffie, 2025-04-06:
> Thank you for the detailed bug report. Can you elaborate if the last release is also affected ?

Based on the git details (`git tag --contains ...` is empty), I'd say: no. I think the PR #483 is too recent for it to be included in V4.1.0 (although the commits are from Nov 2024).

hoffie, 2025-04-06:
Further results after sprinkling in some debug prints:
[`PROGMEM`|https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html] is new to me (I'm not that deep into ESP or Arduino), but `fontSize` and `fontHeight` are declared to be that. It seems like that those are not supposed to be directly referenced as arrays and require pgm_read_byte() instead (?).
For some reason, `fontWidth[0]` ("normal size") seems to crash, while `fontWidth[1]` ("small size") does not. This might explain why it did not show up during testing.

hoffie, 2025-04-06:
I think 1ea211006d787bfdb86d4fc258f8a7bd153a149e (+ 3fd2b248) are other problematic commits (maybe also relevant for #550).

I believe the accesses to `fontWidth` and `fontHeight` are the problem, as they are defined as `PROGMEM`. I'm working on a potential fix.

dbambus, 2025-04-06:
Hey,

I could have a look to the code earliest on Tuesday.    
Then we could elaborate the problem together ;-) till then I m thankful for every hint/solution you can give me or came up with. 

But one remark here the fontsizes shouldn’t be in the transitions. For a cleaner code each mode should have its own unique.h/.hpp or class to interface with. 
Well some To Do here. 🤯 

Cheers 
David

hoffie, 2025-04-06:
> till then I m thankful for every hint/solution you can give me or came up with.

Sure. I've submitted #558 which fixes the issues for me.

> But one remark here the fontsizes shouldn’t be in the transitions. For a cleaner code each mode should have its own unique.h/.hpp or class to interface with.
> Well some To Do here. 🤯

I understand. If possible, I would want to focus on fixing the annoying bug here first and will probably leave the refactoring to you...? :)

PR #558 (2025-04-09): Fix crash in Digital clock and Seconds modes
PR #483 introduced changes which reference the fontWidth and fontHeight variables directly as arrays. However, those two variables are declared as PROGMEM and may therefore only be referenced as addresses which will be read from flash via pgm_read_byte.

The array access seems to cause crashes in certain configurations (and maybe undefined behavior in others?).

This commit changes all fontWidth and fontHeight array accesses to be in line with the previous usage of those variables: The data is read via pgm_read_byte.

Fixes: #557
Related: https://gi [... cut]
commit af332a20 (2025-04-09): Fix crash in Digital clock and Seconds modes

PR #483 introduced changes which reference the fontWidth and fontHeight
variables directly as arrays. However, those two variables are declared
as PROGMEM and may therefore only be referenced as addresses which will
be read from flash via pgm_read_byte.

The array access seems to cause crashes in certain configurations (and
maybe undefined behavior in others?).

This commit changes all fontWidth and fontHeight array accesses to be in
line with the previous usage of those variables: The data is read via
pgm_read_byte.

Fixes: #557
Related: https://g [... cut]
closed 2025-04-09 by dbambus