ESPWortuhr/Multilayout-ESP-Wordclock #560: Helligkeitsregelung mit LDR wegen BH1750-Integration unbrauchbar

D-ominik, 2025-04-09:
Mit #408 wurde eine Integration für den BH1750 ergänzt. Ich verwende bei allen Uhren einen LDR und in Kombination mit der neuen Firmware ist die Helligkeitsregelung vollkommen unbrauchbar, es wird gar nicht mehr geregelt.

@STT65 hast du dein Update mal mit einem LDR getestet? Welche Lux-Werte sollten zu erwarten sein? Ich habe völlig abgedeckt Werte um 2000 Lux herum und mit normaler Raum-Helligkeit 46631 Lux. Letzteres scheint der Maximalwert zu sein, da er auch mit intensiver Beleuchtung nicht weiter steigt.

Ich verwende 5528er LDR mit 10k Widerstand gegen GND.

D-ominik, 2025-04-18:
@dbambus Ich hatte nun einmal Zeit, mir das Problem näher anzusehen, ich denke mit der BH1750-Integration ist für den LDR-Fallback etwas schief gelaufen:

- @STT65 kommentiert im Code einen ADC-Spannungsbereich von 0-1V. Das ist korrekt für den reinen ESP-Chip. Den wird aber wohl niemand für dieses Projekt verwenden, zumal die Doku explizit die Verwendung von Development-Boards vorsieht (z.B. NodeMCU). Die Dev-Boards haben (zumindest alle gängigen) einen Spannungsteiler an den Analog-Pins, um einen Pegel bis Boardspannung (3.3V) zu ermöglichen. Das ist zweckmäßiger und grillt den Eingang nicht so schnell. Weil Stefan seinen Spannungsteiler auf maximal 1 Volt ausgelegt hat, der Port aber 3.3 Volt erlaubt, ist die Auflösung schlecht.
 
Und wenn man kein Dev-Board verwendet, wäre es nicht ratsam, einen LDR ohne Vorwiderstand (wie im vorgesehenen Schaltbild) mit 3.3V zu versorgen, wenn der Eingang nur 1V verträgt, da selbst ein LDR mit einem Hell-Nennwiderstand von mehreren Kilo-Ohm (der Hellwiderstand bezieht sich auf nur 10 Lux) bei sehr hellem Licht (direktes Sonnenlicht, helle Taschenlampe) bis unter 50 Ohm runter geht. Das Ende des Analogeingangs wäre vorprogrammiert.

- In Stefans Kommentar ist ein Spannungsteiler mit einem 220 Ohm Widerstand schematisiert. Den LDR hat er nicht spezifiziert, die üblichen LDR liegen jedoch i.d.R. bei mehreren kOhm bei normaler Raumhelligkeit (ohne direkte Bestrahlung). An einer Wand ohne gegenüberliegendes Fenster ist der Widerstand schnell im Bereich jenseits von 30 kOhm. Das resultiert, bei einer üblichen Lichtsituation im Inneren und dem Anwendungsfall der Uhr, in einer Spannung von 0,3V bis 0,01V an einem Analogeingang der bis 3.3V vorsieht und nur mit 10Bit (0-1023) auflöst. Der Bereich von Dämmerung bis Dunkelheit liegt dann bei 0,01V bis 0,001V (und kleiner). Da ist es nicht überraschend, dass Stefan keine zufriedenstellenden Ergebnisse in dunkler Umgebung erzielte.
 
Mich hat gewundert, wieso er einen Spannungsteiler mit 220 Ohm realisiert hat, wo die Anleitung doch 10k vorgibt. Nach etwas suchen bin ich fündig geworden: in `pics/Schematic.(png|pdf)` ist ein Widerstand mit Farbcode rot-rot-braun, also 220 Ohm abgebildet. Das sollte in 10kOhm geändert werden (am besten auch beschriften). Ich denke es wäre auch sinnvoll einen geeigneten LDR-Typ zu empfehlen, z.B. den 5528.

 
 
Da die Helligkeitsregelung mit Schaltung nach Anleitung (mit 10k Widerstand) gar nicht mehr funktioniert, habe ich den Aufbau mal separat nachgestellt und die drei Messvarianten (BH1750, altes LDR-Schema, neues LDR-Schema) parallel integriert. Jeweils mit den default-Werten für autoBrightOffset, autoBrightSlope, autoLdrBright, autoLdrDark, etc.:
```
BH1750: 1.67 lx, 39 gain
LDR alt: 19 lx, 13 gain
LDR neu: 357.21 lx, 88 gain

BH1750: 2.50 lx, 39 gain
LDR alt: 27 lx, 16 gain
LDR neu: 683.49 lx, 100 gain

BH1750: 5.83 lx, 39 gain
LDR alt: 49 lx, 24 gain
LDR neu: 1986.40 lx, 100 gain

BH1750: 15.83 lx, 41 gain
LDR alt: 75 lx, 34 gain
LDR neu: 4406.32 lx, 100 gain

BH1750: 32.50 lx, 43 gain
LDR alt: 117 lx, 49 gain
LDR neu: 10236.60 lx, 100 gain

BH1750: 43.33 lx, 44 gain
LDR alt: 131 lx, 54 gain
LDR neu: 12715.65 lx, 100 gain

BH1750: 452.50 lx, 100 gain
LDR alt: 231 lx, 91 gain
LDR neu: 38130.87 lx, 100 gain
```
Bis auf den Bereich voller Abdunklung wird mit der neuen Logik mit LDR immer maximale Helligkeit ausgesteuert. Mit einem 220 Ohm Widerstand gehen die "Lux"-Werte nicht so durch die Decke, erwartungsgemäß sind die Werte bei wenig Helligkeit dann jedoch recht "festgenagelt".

Der "gain" bei wenig Helligkeit ist auch beim BH1750 recht hoch und im unteren Helligkeitsbereich zu statisch. Das ließe sich mit den Konfigurationswerten etwas optimieren. Generell ist es nicht einfach, mit der Kombination "Mindesthelligkeit" und "Steilheit" zielführende Resultate zu erreichen. Es braucht viele Versuche bei verschiedenen Helligkeiten. Wenn man dann noch bestimmte Maximalwerte nicht überschreiten möchte... Im Prinzip war die alte Lösung, die Minimal- und die Maximalhelligkeit zu definieren, leichter verständlich, die gewünschten Resultate wurden schneller erzielt (keine langwierigen Versuche) und, wie man am Verlauf von "LDR alt" sieht, auch zweckdienlicher.

D-ominik, 2025-04-18:
Ich habe die LDR-Auswertung überarbeitet und eine klassische Lux-Umrechnung gewählt. Innerhalb der Helligkeitsbereiche von  Raumlicht erhält man sehr brauchbare Werte (für den Einsatzzweck, ist ja kein Messgerät). Das Ganze ohne statische Kurvenauswertung. Die Werte des Spannungsteilers sind parametrisiert und daher bei abweichendem Setup anpassbar.
```
BH1750:     1.67 lx 
LDR alt:    10 lx
LDR neu:    134.94 lx
LDR rework: 2 lx

BH1750:     4.17 lx 
LDR alt:    22 lx
LDR neu:    479.61 lx
LDR rework: 6 lx

BH1750:     70.00 lx 
LDR alt:    135 lx
LDR neu:    13521.21 lx
LDR rework: 74 lx

BH1750:     90.83 lx 
LDR alt:    156 lx
LDR neu:    17863.89 lx
LDR rework: 104 lx

BH1750:     146.67 lx 
LDR alt:    176 lx
LDR neu:    22746.74 lx
LDR rework: 149 lx

BH1750:     224.17 lx 
LDR alt:    196 lx
LDR neu:    27871.67 lx
LDR rework: 221 lx

BH1750:     465.83 lx 
LDR alt:    221 lx
LDR neu:    35357.46 lx
LDR rework: 439 lx

BH1750:     595.00 lx 
LDR alt:    231 lx
LDR neu:    38211.65 lx
LDR rework: 622 lx

BH1750:     617.50 lx 
LDR alt:    230 lx
LDR neu:    38211.65 lx
LDR rework: 622 lx
```

Phrunky, 2025-04-21:
Hallo D-ominik,

vielen Dank für Deine Erläuterungen. Ich habe ebenso versucht, die Helligkeitsanpassung für mich zu verbessern und habe nur unwesentliche Fortschritte gemacht - allerdings habe ich die Geschichte bisher nur hardwareseitig betrachtet. 
Vorab: bitte entschuldige, wenn ich unqualifiziert frage, aber kannst Du ausführen, was Du unter den Ausführungen "LDR alt", "LDR neu" und "LDR rework" meinst? "LDR alt" entspricht wohl dem 220 Ohm-Widerstand "LDR neu" vermutlich dem 10k Ohm. Und ich verstehe auch noch nicht, ob und wo es im Programmcode eine Umsetzungstabelle für die unterschiedlichen Belichtungsgrade gibt.

Sorry für meine Fragen, aber die Doku gibt da auch nicht viel her.

PS Ich bin die Geschichte bisher noch nicht mit dem BH1750 angegangen.

Phrunky, 2025-04-25:
Vielen, vielen Dank für die ausführliche Information und das eingebrachte Engagement

closed 2025-05-22 by D-ominik