ESPWortuhr/Multilayout-ESP-Wordclock #578: Richtige Einstellungen für panbachi mit Minutenanzeige

skar-cgn, 2025-04-20:
Hallo, 
danke für das Projekt. Aufgrund der Möglichkeiten bin ich von panbachi auf diese Software umgestiegen. 
Ich bekomme es aber nicht hin die Minutenanzeige passend in einzurichten. Die Minuten sind ja eigentlich mittig: 
https://github.com/panbachi/wordclock/blob/master/graphics/plate/DE_de_s1.jpg

Egal was ich einstelle, die vier Minutensymbole werden nie getroffen. 

Hat das jemand schon mal gemacht?

Grüße
Ralf

obkram, 2025-05-12:
Hallo liebes Entwicklerteam,

Das gleiche Problem habe ich auch. Ich möchte auch gerne mit meiner WC von Panbachi auf dieses Projekt umsteigen. Leider gibt die folgenden Problem. Es wäre prima wenn sich jemand die Mühe machen könnte das nachzulegen.

1. Die Panbachi Variante beginnt mit der Zählweise her unten rechts. Dieses Projekt hier beginnt oben links. Das bekommt man mit der Einstellung Wortuhr vertikal und Wortuhr horizontal spiegeln super hin. Leider bleiben die Minuten immer an der gleichen Stelle und überschreiben den Text " Es ist ..."

2. Die LEDs werden ( bei der Zählweise von oben links ) im Minuten Anzeige Modus mit 7 LEDs auf 103 | 105 | 107 | 109 benötigt. Derzeit sind diese in diesem Modus für die Minutenanzeige leider auf den LEDs  102 | 104 | 106 | 108 angelegt, was nicht zum Layout der WC passt. Ein umziehen von diesem Projekt auf eine bestehende Uhr ist somit nicht möglich.

3. Die 4 Minuten LED Variante macht keinen Sinn wenn das Layout in der Horizontalen aus 11 Zeichen besteht, da es sonst unsymmetrisch ausschaut. Die vier Minuten LED Variante macht nur Sinn wenn diese für die Ecken genutzt wird, was dann bei der Panbachi Zählweise in der Matrix 11x11 wie auch bei diesem Projekt die LEDs 112 | 113 | 114 | 115 sein müsste. Diese müssten dann auch nicht beim Horizontalen / Vertikalen Spiegeln verschoben werden, dann sie immer ausserhalb des aktiven Zifferblattes wären.

dbambus, 2025-05-21:
Hallo @obkram und @skar-cgn, 

vielen Dank für das Issue. Ich wusste selbst nicht über die andere Bauweise dieses Projektes bescheid und hatte es so genommen wie es ist. 

anbei habe ich mal den Code der DE11x11.v3 geändert: 

```
#pragma once

#include "DE11x11.hpp"

/*
 *           Layout Front
 *                COL
 *       X 9 8 7 6 5 4 3 2 1 0
 * ROW + - - - - - - - - - - -
 *  0  | E S K I S T L F Ü N F
 *  1  | Z E H N Z W A N Z I G
 *  2  | D R E I V I E R T E L
 *  3  | T G N A C H V O R J M
 *  4  | H A L B X Z W Ö L F P
 *  5  | Z W E I N S I E B E N
 *  6  | K D R E I R H F Ü N F
 *  7  | E L F N E U N V I E R
 *  8  | W A C H T Z E H N R S
 *  9  | B S E C H S F M U H R
 *  X  | E V * X * N * S * P I
 */

class De11x11V3_t : public De11x11_t {
public:
    virtual const void getMinuteArray(uint16_t *returnArr,
                                      uint8_t col) override {

        uint16_t numPixelsWordMatrix = rowsWordMatrix() * colsWordMatrix();

        if (G.buildTypeDef == BuildTypeDef::DoubleResM1) {
            numPixelsWordMatrix = rowsWordMatrix() * (colsWordMatrix() * 2 - 1);
        }

        for (uint8_t i = 0; i < 4; i++) {
            switch (col) {
            case 0:                         // LEDs for "LED4x" minute display
                returnArr[i] = 3 + (i * 2); // 3,5,7,9
                break;

            case 1:                         // LEDs for "LED7x" minute display
                returnArr[i] = 113 + (i * 2); // 3,5,7,9
                break;

            case 2: // LEDs für "Corners" type minute display
                returnArr[i] = numPixelsWordMatrix + I;
                break;

            default:
                break;
            }
        }
    };

    //------------------------------------------------------------------------------

    void show(FrontWord word) override {
        switch (word) {

        case FrontWord::es_ist:
            setFrontMatrixWord(0, 9, 10);
            setFrontMatrixWord(0, 5, 7);
            break;

        case FrontWord::viertel:
            setFrontMatrixWord(2, 0, 6);
            break;

        case FrontWord::dreiviertel:
            setFrontMatrixWord(2, 0, 10);
            break;

        case FrontWord::min_5:
            setFrontMatrixWord(0, 0, 3);
            break;

        case FrontWord::min_10:
            setFrontMatrixWord(1, 7, 10);
            break;

        case FrontWord::min_20:
            setFrontMatrixWord(1, 0, 6);
            break;

        case FrontWord::halb:
            setFrontMatrixWord(4, 7, 10);
            break;

        case FrontWord::eins:
            setFrontMatrixWord(5, 5, 8);
            break;

        case FrontWord::nach:
        case FrontWord::v_nach:
            setFrontMatrixWord(3, 5, 8);
            break;

        case FrontWord::vor:
        case FrontWord::v_vor:
            setFrontMatrixWord(3, 2, 4);
        case FrontWord::uhr:
            setFrontMatrixWord(9, 0, 2);
            break;

        case FrontWord::hour_1:
            setFrontMatrixWord(5, 6, 8);
            break;

        case FrontWord::hour_2:
            setFrontMatrixWord(5, 7, 10);
            break;

        case FrontWord::hour_3:
            setFrontMatrixWord(6, 6, 9);
            break;

        case FrontWord::hour_4:
            setFrontMatrixWord(7, 0, 3);
            break;

        case FrontWord::hour_5:
            setFrontMatrixWord(6, 0, 3);
            break;

        case FrontWord::hour_6:
            setFrontMatrixWord(9, 5, 9);
            break;

        case FrontWord::hour_7:
            setFrontMatrixWord(5, 0, 5);
            break;

        case FrontWord::hour_8:
            setFrontMatrixWord(8, 6, 9);
            break;

        case FrontWord::hour_9:
            setFrontMatrixWord(7, 4, 7);
            break;

        case FrontWord::hour_10:
            setFrontMatrixWord(8, 2, 5);
            break;

        case FrontWord::hour_11:
            setFrontMatrixWord(7, 8, 10);
            break;

        case FrontWord::hour_12:
            setFrontMatrixWord(4, 1, 5);
            break;

        case FrontWord::funkt
            break;

        default:
            break;
        };
    };
};

De11x11V3_t _de11x11V3;
```

Könnt ihr das mal testen ? 

Mit freundlichen Grüßen
David

obkram, 2025-05-21:
@dbambus 
Vielen Dank das du dir die Mühe gemacht hast das anzupassen. Der Code sieht schon mal gut aus.

@skar-cgn 
Ich habe im Moment nur die Möglichkeit mit fertigen Release OTA .bin‘s zu arbeiten. Hast du einen aktiven Rechner um die Codezeilen für dbambus bei dir zu ändern und zu testen?

obkram, 2025-05-21:
Hallo @dbambus,

ich habe es auf zwei ESPs getestet. Das Ergebnis bleibt das Gleiche. Die Minuten LEDs sind bei dem Modus mit 7 LEDs um eins versetzt und werden beim horizontalen und vertikalen Spiegeln nicht berücksichtigt . Die Releaseversion ist allerdings nicht auf die 4.1.2 gewechselt sonder zeigt weiterhin 4.1.0 an. 

[image]

obkram, 2025-05-22:
Kleiner Hinweis noch: bei der Panbachi Variante ist der Daten Einspeisepunkt in der untersten Reihe rechts auf dem „I“.

obkram, 2025-05-24:
Vielleicht könnte man ja auch die feste Hinterlegung der Minutenpositionen in den einzelnen Layouts weglassen und im Webinterface ein Feld zur Angabe der ersten Minuten LED anhand der LED Zählweise hinzufügen, von welcher aus dann 4 oder 7 LED weitergezählt werden.

dbambus, 2025-06-08:
Hallo @obkram, 

Ichhabe nochmal eine neue Version erstellt. Kannst du die mal testen ? 

Mit freundlichen Grüßen
David

[4.2.0_Test.bin.zip](https://github.com/user-attachments/files/20646125/4.2.0_Test.bin.zip)

obkram, 2025-06-08:
Hi @dbambus ,
vielen Dank für deine Mühe. Ich habs nochmal getestet. Das Ergebnis bleibt das Gleiche ( siehe Bild ). Die Minuten sind um eine Position verschoben. Unverändert ist auch das egal wie gespiegelt wird ( ebenfalls der neue Regler Horizontales und Vertikales Layout drehen ) die Minuten immer auf den gleichen vier LEDs im Streifen angezeigt werden. Nur Minutenzählrichtung hat eine Auswirkung auf die Leds.

[image]

Andreas-78, 2025-06-09:
Jetzt weiß ich wieder glaube die hatte das Problem mit fehlenden „zwanzig“ weshalb ich sie dann wieder umgebaut habe.

dbambus, 2025-06-10:
> vielen Dank für deine Mühe. Ich habs nochmal getestet. Das Ergebnis bleibt das Gleiche ( siehe Bild ). Die Minuten sind um eine Position verschoben. Unverändert ist auch das egal wie gespiegelt wird [...]

Hey @obkram 

mittlerweile bin ich etwas verwirrt. ich habe nur die LED4x verändert und die LEDs der Minuten darin hard auf 3, 5, 7, 9 gecoded. kannst du das nochmal testen ob es jetzt geht ? 

Mit freundlichen Grüßen
David

skar-cgn, 2025-06-10:
So, jetzt bin ich richtig verwirrt. 
Mit dem letzten Pre Release geht es nicht, dann sind die Minuten in der unteren Reihe um eine LED nach links verschoben. 
Mit der Version 4.2.0_Test.bin.zip geht es aber. Mit der Version 4.1.2 wiederrum nicht. 

In meiner Version ist LED Anschluss von vorne gesehen unten rechts. 

Grüße
Skar
Foto: [image]
Einstellungen: [image]

obkram, 2025-06-12:
Hallo @dbambus ,

anbei nochmal ein Foto als Nachweis das bei der 4.1.2 alles klappt wie gewohnt. Ist meine Erste Uhr vor vielen Jahren gewesen, deshalb ist hier ein kleiner Fehler drin. Ich habe mich damals verlasert. Die guten Uhren habe ich alle verschenkt und nur diese behalten.

Wenn man das ganze Projekt das erste Mal auf der Uhr sieht und nicht nur auf einem LED Streifen dann ist man ersteinmal erschlagen wie cool die ganzen Sonderfunktionen ausschauen. Ich habe das grinsen nicht mehr aus dem Gesicht bekommen. Da habt ihr wirklich eine ganz tolles Projekt auf die Beine gestellt. Großes Lob und einen riesigen Dank an das ganze Team.

Zitat @dbambus : "Für die „Panbachi“-Variante der Uhr wird es in Zukunft keine Auswahl geben, da hier nur die eine Layout-Variante unterstützt werden soll."

Wenn es keinen Aufwand macht dann würde ich bei der Panbachi Variante die Auswahl " Ecken " die in der 4.2.0 drin war beibehalten, da es auch diese Variante bei Panbachi gibt. In diesem Fall hattet ihr das schon richtig angelegt das die vier Minuten LEDs einfach an die 120ste LED folgend angehängt werden.

Wie kann ich mir denn die Code Variante Minutes::InWords vorstellen? Ich habe da evtl. ein falsches Bild zu im Kopf. Werden die Sekunden dann zusätzlich in die Matrix als Wort mit eingepflegt?

Ich habe beim Flaschen gesehen das im Editor 5 mögliche Geburtagstermine angezeigt wurden. Werden diese noch für das Webinterface editierter gemacht?

[image]

dbambus, 2026-10-23:
Hallo @skar-cgn und @obkram,

vielen Dank für das Issue und eure Geduld beim Testen der ganzen Testversionen, und sorry für das Hin und Her mit LED4x und LED7x. Für die Variante „11 × 11 (panbachi)“ ist es jetzt fest so gelöst wie besprochen: Die vier Minutenpunkte sind Teil der Buchstabenmatrix (Minutenanzeigemodus „In Worten“) und liegen auf den Sternchen in der untersten Reihe. Weil sie zur Matrix gehören, werden sie beim horizontalen und vertikalen Spiegeln automatisch mitgespiegelt und landen bei Einspeisung unten rechts auf den richtigen LEDs.

Zu euren offenen Fragen:
- Die Variante „Ecken“ gibt es ab 5.0 nicht mehr, die 11x11-Layouts nutzen die normalen Minuten-LEDs.
- Die Geburtstage lassen sich jetzt im Webinterface eintragen.

Enthalten im Release Candidate V5.0.0-rc.1:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V5.0.0-rc.1

Achtung: Mit dem Update werden alle Einstellungen zurückgesetzt.

Ich schließe das Issue. Falls die Minuten bei euch noch versetzt sind, öffnet es gerne wieder.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus

obkram, 2026-10-24:
@dbambus Hi David, danke für die Rückmeldung. Unglaublich das du das Thema nach so langer Zeit noch auf dem Schirm hattest. Ich habe eben mal das aktuellste Release mal auf meiner Panbachi Uhr aufgespielt. Leider ist hier noch ein kleiner Fehlerteufel drin. Die Position der Minuten ist korrekt. Leider bleiben die LEDs beim rauf zählen nicht stehen sondern es wird immer nur die aktuelle Minuten-LED angezeigt. Anbei sende ich dir zum besseren Verständnis Bilder der Uhrzeit 13:22h,13:23h und 13:24h. Die vergangenen Minuten sollten aber stehen bleiben bis zum nächsten 5 Minutenwechsel. Es wäre nett wenn du da nochmal drüber schauen könntest. Bei Rückfragen melde dich gerne bei mir.

[image]
[image]
[image]

obkram, 2026-10-24:
@dbambus Vielleicht kannst du das Ticket erstmal wieder öffnen 😉. Danke Dir

obkram, 2026-10-24:
@dbambus Noch ein kleiner Hinweis, der Funktionsmodus Feuer wird in der aktuellen Version nicht mit gespiegelt. Alle anderen Modis passen

dbambus, 2026-10-24:
Hallo @obkram (und @skar-cgn),

vielen Dank fürs Testen und die genauen Rückmeldungen, besonders für die Fotos. Beide Fehler sind jetzt behoben:

- **Minutenpunkte bei der Panbachi-Variante:** Die Punkte bleiben jetzt beim Hochzählen stehen. Bei Minute 3 leuchten also die Punkte 1 bis 3, bis zum nächsten 5-Minuten-Wechsel. Commit: https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/commit/3da3174b95ac8a525059c68bc333a1f6e3d98ac2
- **Feuer wurde nicht gespiegelt:** Der Modus Feuer folgt jetzt dem horizontalen und vertikalen Spiegeln. Den Modus Regenbogen-Zyklus, der das gleiche Problem hatte, habe ich gleich mit angepasst. Commit: https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/commit/98ddc4e98e8bfa48e3aba74ad3f07abb0740e54f

Zum Testen musst du nicht selbst kompilieren. Der Nightly-Build läuft heute Nacht und enthält dann beide Fixes:
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/actions/workflows/nightly.yml

Es wäre super, wenn du das morgen auf deiner Panbachi-Uhr ausprobieren könntest. Falls noch etwas nicht passt, melde dich gerne hier.

Mit freundlichen Grüßen
David

obkram, 2026-10-25:
@dbambus Hi David, ich weiß leider nicht was ich mit der Nightly anstellen muss. Die enthaltene Zip Datei kann ich leider nicht entpacken. Ich muss leider mit dem Test bis zur nächsten Beta Version warten.

@skar-cgn Es sei denn du könntest das vorab testen.

obkram, 2026-10-27:
@dbambus Hi David, ich habe das Release von heute Morgen aufgespielt und kann die bestätigen das nun alles korrekt funktioniert. Vielen Dank dafür ☺️!