ESPWortuhr/Multilayout-ESP-Wordclock #583: Sekunden Anzeige möglich?

Andreas-78, 2025-04-25:
Hallo, könnte mir jemand sagen bei welchem Layout oder was man noch verbauen muss damit die Sekunden dargestellt werden.

Ich habe eine Uhr gebaut in der Matrix 10x11 jedoch egal welchen Uhrentyp ich auswähle ich bekomme weder Punkt noch Sektor angezeigt.

Es ist ja eigentlich nur im 10x11 Alternativer Rahmen auswählbar jedoch bekomme ich nichts von Sekunden angezeigt.Muss ich demCode noch etwas hinzufügen?

Danke schonmal

dbambus, 2025-04-25:
Hey @Andreas-78,

leider ist da bei der software auf 4.1.0 etwas schief gelaufen, irgendwie sehr viel und deswegen ist die Funktion unvollständig drin. Probiere doch mal V4.0.0 in der release Version, da sollte es funktionieren. 

Der Aufbau ist dabei wie folgt. zuseht kommen die 110 LEDs der Wordclock dann die 4 LEDs der Minuten und diese sind gefolgt von 48 LEDs der Sekunden (12 Pro Seite der Uhr, beginnend oben Links über dem ES). 

Eine entsprechende Doku solle hoffentlich bald folgen. 

Mit freundlichen Grüßen
David

Andreas-78, 2025-04-25:
Danke für die schnelle Erklärung.
Ich bin davon ausgegangen das die LED aus der Matrix genutzt werden können für die Minuten.
Also quasi wie die Hintergrund Beleuchtung nur das eben immer nur eine LED angeht und um eins weiterspringt.

Andreas-78, 2025-04-26:
Hi @dbambus gibt es da schon eine fertige SVG Datei mit den 12x4 Ausschnitten für die Sekunden?
Oder ein Bild wie das aussehen könnte?

dbambus, 2025-04-26:
Hallo @Andreas-78,

Ich bin grad noch unterwegs, sonst könnte ich ein Modell Foto machen. 

Ich habe jetzt nur das aus der Readme.md 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pics/SecondsInFrame.gif

Mit freundlichen Grüßen
David

Andreas-78, 2025-04-26:
Hallo,

Ja jetzt erkenn ich es da ist quasi der Hintergrund Beleuchtet

dbambus, 2025-04-26:
Genau,

Der geht dann entweder als Punkt, Sektor, … eigentlich will ich auch in Zukunft eine Ambientlight Funktion implementieren. ;-) 

Mit freundlichen Grüßen
David

Andreas-78, 2025-04-26:
Hallo,

ja verstehe.
Mein Ansatz war ein anderer.
Da es ein Ikea Rahmen ist hängt der an der Wand und so ist das mit dem Ambilight nicht möglich.
Deshalb würde mir ein blinkender Punkt reichen.
Alternativ würde ich die Punkte wie für die Minuten auf die Seiten verteilen, bin grad noch dran und kann dir ja mal die SVG senden

Mit freundlichen Grüßen
Andi

Andreas-78, 2025-05-09:
Hi,

hab sie endlich fertig bekommen.
Läuft wunderbar mit der Version 4.0.0.
Da ich nur noch einen Lichtsensor TSL 2561 Lichtsensor da habe die Frage gibt es eine Möglichkeit diesen einzubinden?

Lg Andreas

> 
> 
> dbambus
>  left a comment 
> (ESPWortuhr/Multilayout-ESP-Wordclock#583)
> Genau,
> 
> Der geht dann entweder als Punkt, Sektor, … eigentlich will ich auch in Zukunft eine Ambientlight Funktion implementieren. ;-)
> 
> Mit freundlichen Grüßen
> David
￼

dbambus, 2025-05-21:
Hey @Andreas-78, 

das sollte grundsätzlich möglich sein, in der `include/clockwork.hpp` Zeile 10 -70 ist der code zur Helligkeit zu finden. hier kann man auch einen anderen Sensor Einpflegen. 

Ich hoffe das konnte helfen.

Mit freundlichen Grüßen
David

closed 2025-05-21 by dbambus