ESPWortuhr/Multilayout-ESP-Wordclock #586: Colored Word

Jayrox3000, 2025-04-27:
If my light sensor turns off the Leds, after turning them back on they are all white and don't have colored Words anymore. To fix it i have to turn Colored Words back off and on again, but this fixes only temporary until the light sensor turns the leds off and on again.

BastiMu, 2025-05-07:
Bei mir haben die Wörter nach einem Zeitwechsel wieder die richtige Farbe. Ist das bei dir auch so?

dbambus, 2025-05-21:
Hey @Jayrox3000 und @BastiMu, 

vielen Dank für das Issue. Ich muss jetzt aber mal laborieren. 

Die Option "Farbige Wörter"  ist das AN? Ja/Nein
Ist ein Helligkeitssensor verbaut, also nutzt du die automatische Einstellung oder die Manuelle? Ja/Nein

Je nachdem was aktiviert ist, kann eine andere Fehlerquelle vorhanden sein. 

Mit freundlichen Grüßen
David

BastiMu, 2025-05-21:
Hallo @dbambus 

- Farbige Wörter ist an
- Helligkeitssensor war bei mir nicht verbaut und Einstellung auf Manuell

dbambus, 2026-10-23:
Hallo @Jayrox3000 und @BastiMu,

vielen Dank für das Issue und fürs Nachstellen. Ich konnte den Fehler jetzt genau eingrenzen: Bei „Bunte Wörter“ (in 5.0 „Zufall pro Wort“) steckte die Farbe eines Wortes nur in seinen RGB-Werten. Bei Helligkeit 0 werden alle Wörter schwarz und verlieren dabei ihren Farbton. Wird es wieder hell, kommt Weiß heraus – bis zum nächsten Wortwechsel, genau wie ihr es beobachtet habt. Bei sehr niedriger Helligkeit (z. B. über den Lichtsensor) haben sich die Farben außerdem leicht verschoben, und ohne Übergang wurden sie in 5.0 sogar bei jeder Helligkeitsänderung neu gewürfelt.

Mit #829 merkt sich die Uhr jetzt die Farbe jedes Wortes separat und setzt sie nach einer Helligkeitsänderung wieder ein. Die Wörter behalten ihre Farben also durch 0 % und zurück, bei sehr wenig Helligkeit und bei Minutenwechseln – mit und ohne Übergang. Neue Farben gibt es nur noch, wenn neue Wörter erscheinen. Ich habe das auf einer echten Uhr getestet.

Der Fix kommt mit der nächsten Version. Ich schließe das Issue; falls die Wörter danach doch noch weiß werden, öffnet es gerne wieder.

Mit freundlichen Grüßen
David

closed 2026-10-23 by dbambus