ESPWortuhr/Multilayout-ESP-Wordclock #624: ESP32-C3 (SuperMini) - Boot Loop

electrontube, 2025-07-27:
Ich habe jetzt einige Zeit mit verschiedenen ESP32-C3 verbracht, jedoch leider ohne Erfolg.
Die ESP32-C3 sind in Ordnung und laufen Fehlerfrei mit anderer Software. 
Umgekehrt läuft die Uhrensoftware problemlos auf einem WeMos D1.
Es muss also ein Problem mit der Konfiguration für den ESP32C3 sein.
Compile und Upload laufen fehlerfrei, aber dann endet der ESP32C3 in einer BootLoop

Hat jemand ein funktionierendes Beispiel für den ESP32C3 (SuperMini)?

Miklesoft, 2025-08-21:
Hi, bei mir funktionierts mit folgenden Einträgen in platfom.ini
[env:esp32c3dev]
platform = espressif32 @ 6.9.0
board = esp32-c3-devkitc-02
board_build.partitions = partitions_singleapp_large.csv
framework = arduino
upload_speed = 921600
monitor_speed = 115200
build_flags = 
    -Os
    -ffunction-sections
    -fdata-sections
    -Wl,--gc-sections
lib_deps =
    makuna/NeoPixelBus@^2.7.6
    bblanchon/ArduinoJson@^6.17.2
    links2004/WebSockets@^2.4.1
    adafruit/RTClib@^1.11.2
    knolleary/PubSubClient@^2.8.0
    https://github.com/tzapu/WiFiManager#v2.0.17
    claws/BH1750@^1.3.0
extra_scripts = pre:extra_scripts.py

electrontube, 2025-08-25:
Ich habe mal ein wenig experimentiert und bin zu folgendem Ergebnis gekommen:

partitions-esp32c3.csv

```
# Name,      Type, SubType, Offset,   Size
nvs,         data, nvs,     0x9000,   0x5000
otadata,     data, ota,     0xe000,   0x2000
app0,        app,  ota_0,   0x10000,  0x1C0000
app1,        app,  ota_1,   0x1D0000, 0x1C0000
coredump,    data, coredump,0x390000, 0x10000
phy,         data, phy,     0x3A0000, 0x1000

```
config.h
```
//--------------------------------------------------------------------------
// PIN Configuration
//--------------------------------------------------------------------------
#define LED_PIN 4       // Use direct pin number
#define SDA_PIN_ESP32 8 // Use direct pin number
#define SCL_PIN_ESP32 9 // Use direct pin number

```
platformio.ini

```
[platformio]
default_envs = ESP32C3

[env:ESP32C3]
platform = espressif32 @ 6.9.0
board = esp32-c3-devkitc-02
board_build.partitions = partitions-esp32c3.csv
framework = arduino
upload_speed = 921600
monitor_speed = 115200
build_flags = 
    -Os
    -ffunction-sections
    -fdata-sections
    -Wl,--gc-sections
lib_deps =
    makuna/NeoPixelBus@^2.7.6
    bblanchon/ArduinoJson@^6.17.2
    links2004/WebSockets@^2.4.1
    adafruit/RTClib@^1.11.2
    knolleary/PubSubClient@^2.8.0
    https://github.com/tzapu/WiFiManager#b67b7828b9192fcbbc3f723c15b2bfa83e6a22fd
    claws/BH1750@^1.3.0
extra_scripts = pre:extra_scripts.py

```

Damit funktioniert auch OTA Update.
@dbambus
Kannst Du Dir das bitte mal anschauen und wenn ok so ins nächste Release übernehmen?

Vielen Dank!

( z.K. @obkram )

Gruß
R.

electrontube, 2025-08-26:
> > Ich habe mal ein wenig mit dem ESP32-C3 SuperMini experimentiert und bin zu folgendem Ergebnis gekommen:
> > partitions-esp32c3.csv
> > ```
> > # Name,      Type, SubType, Offset,   Size
> > nvs,         data, nvs,     0x9000,   0x5000
> > otadata,     data, ota,     0xe000,   0x2000
> > app0,        app,  ota_0,   0x10000,  0x1C0000
> > app1,        app,  ota_1,   0x1D0000, 0x1C0000
> > coredump,    data, coredump,0x390000, 0x10000
> > phy,         data, phy,     0x3A0000, 0x1000
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > config.h
> > ```
> > //--------------------------------------------------------------------------
> > // PIN Configuration
> > //--------------------------------------------------------------------------
> > #define LED_PIN 4       // Use direct pin number
> > #define SDA_PIN_ESP32 8 // Use direct pin number
> > #define SCL_PIN_ESP32 9 // Use direct pin number
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > platformio.ini
> > ```
> > [platformio]
> > default_envs = ESP32C3
> > 
> > [env:ESP32C3]
> > platform = espressif32 @ 6.9.0
> > board = esp32-c3-devkitc-02
> > board_build.partitions = partitions-esp32c3.csv
> > framework = arduino
> > upload_speed = 921600
> > monitor_speed = 115200
> > build_flags = 
> >     -Os
> >     -ffunction-sections
> >     -fdata-sections
> >     -Wl,--gc-sections
> > lib_deps =
> >     makuna/NeoPixelBus@^2.7.6
> >     bblanchon/ArduinoJson@^6.17.2
> >     links2004/WebSockets@^2.4.1
> >     adafruit/RTClib@^1.11.2
> >     knolleary/PubSubClient@^2.8.0
> >     https://github.com/tzapu/WiFiManager#b67b7828b9192fcbbc3f723c15b2bfa83e6a22fd
> >     claws/BH1750@^1.3.0
> > extra_scripts = pre:extra_scripts.py
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Damit funktioniert auch OTA Update. [@dbambus](https://github.com/dbambus) Kannst Du Dir das bitte mal anschauen und wenn ok so ins nächste Release übernehmen?
> > Vielen Dank!
> > ( z.K. [@obkram](https://github.com/obkram) )
> > Gruß R.
> > P.S. wenn das OK der Experten da ist, kann das hier geschlossen werden.
> 
> Hat bei mit funktioniert! Danke!

@dbambus ?